mavlink/qgroundcontrol #11815: VideoReceiverLog: GStreamer error: Could not initialize window system

nested-slk, 2024-08-28:
## Expected Behavior
The app doesn't open the video stream if I build the app. On the daily version, video stream works.
[image]

## Current Behavior
`VideoReceiverLog: GStreamer error: Could not initialize window system`

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Build from sources
2. Run gstreamer `gst-launch-1.0 --verbose     videotestsrc !     x264enc !     h264parse !     rtph264pay         config-interval=10         pt=96 !     udpsink         host=127.0.0.1         port=5600`
3. Run qgroundcontrol  

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04, **Wayland**
- QGC Version: v4.3.0-816-g7c80109e0
- QGC build: self-built from source

## Detailed Description
I used this guide https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/
I tried disabling x11 macros in CMake, but it didn't help
[image]

## Log Files and Screenshots
```
qgc.qgcapplication: Settings location "/home/sky/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nqgc.*.debug=false\nGStreamerLog.debug=true\nGStreamerAPILog.debug=true\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
GStreamerLog: Failed to change ranking of feature. Featuer does not exist: bcmdec
MAVLinkLogManagerLog: MAVLink logs directory: "/home/sky/Documents/QGroundControl Daily/Logs"
VideoReceiverLog: GStreamer error: Could not initialize window system
VideoReceiverLog: GStreamer error: Could not initialize window system
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
qgc.positionmanager.positionmanager: QGCPositionManager::_setPositionSource(QGCPositionManager::QGCPositionSource)::<lambda(QGeoPositionInfoSource::Error)> QGeoPositionInfoSource::AccessError
VideoReceiverLog: GStreamer error: Could not initialize window system
```

HTRamsey, 2024-08-28:
So you're saying daily works but not when you build it yourself?

nested-slk, 2024-08-29:
> So you're saying daily works but not when you build it yourself?

Yes, you are right.

doquangquan93, 2024-08-30:
I have same problem :(

HTRamsey, 2024-09-04:
Hoping to resolve with #11806

doquangquan93, 2024-09-05:
> Hoping to resolve with #11806

That's great. It works perfectly . Thank so much

closed 2024-10-30 by HTRamsey