mavlink/qgroundcontrol #11816: The joystick does not connect to QgroundControl

WolfbertFX, 2024-08-28:
The Radio Master TX 12 is not connected to the QgroundControl in the daily version. This problem is only on Windows, the build for macOS works correctly. In QT console it looks like this:

[E] at C:\Users\wolfbertfx\Desktop\qgroundcontrol\src\Joystick\JoystickSDL.cc:61 - "	 libsdl failed opening joystick Radiomaster TX12 Joystick error: IDirectInputDevice8::SetCooperativeLevel() DirectX error 0x80070006"

## System Information
- Operating System: [Windows 10 Pro 21H2]
- QT Profile: [QT 6.6.3 MSVC2019 64bit]
- QGroundControl: [Daily Version v4.3.0-818-gd105ff6eb 64bit]
- Сompiler: Microsoft Visual C++ Compiler 17.11.35219.272 (amd64)
- CMake: [3.29.3 (QT)]

HTRamsey, 2024-08-28:
Can you check a previous release? If this is specifically a windows SDL issue that's hard to fix

WolfbertFX, 2024-08-30:
in build 4.4.1 (Release) the joystick works correctly
[image]

HTRamsey, 2024-08-30:
You could try building with a different SDL version, like perhaps the one that 4.4.1 uses and see if that works

WolfbertFX, 2024-09-05:
1) I checked a similar example at the link above, but the Radio Master TX 12 joystick simply does not have the settings described there.

2) In the daily version of QGC, SDL version 2.30.2 is used. In version 4.4.1 of QGC, SDL version 2.0.20 is used. I tried to change the version in BuildSdl2.cmake, but the project did not compile. I haven’t found a way to fix it yet(

Build output:
SDL2d.lib (SDL_dynapi.c.obj) : error LNK2019: unresolved external symbol SDL_RegisterApp_REAL referenced in function initialize_jumptable.
SDL2d.lib (SDL_dynapi.c.obj) : error LNK2019: nresolved external symbol SDL_UnregisterApp_REAL referenced in function initialize_jumptable.
SDL2d.lib (SDL_dynapi.c.obj) : error LNK2019: error LNK2019: unresolved external symbol SDL_Direct3D9GetAdapterIndex_REAL referenced in function initialize_jumptable.
SDL2d.lib (SDL_dynapi.c.obj) : error LNK2019: error LNK2019: unresolved external symbol SDL_DXGIGetOutputInfo_REAL referenced in function initialize_jumptable.
SDL2d.lib (SDL_dynapi.c.obj) : error LNK2019: error LNK2019: unresolved external symbol SDL_SetWindowsMessageHook_REAL referenced in function initialize_jumptable.
SDL2d.lib (SDL_dinputjoystick.c.obj) : error LNK2001: unresolved external symbol SDL_HelperWindow
QGroundControl.exe : fatal error LNK1120
ninja: build stopped: subcommand failed.

hyeonseung89, 2024-10-29:
I'm experiencing the same issue. 

"libsdl failed opening joystick Radiomaster TX16S Joystick error: IDirectInputDevice8::SetCooperativeLevel() DirectX error 0x80070006"

The sdl version is 2.30.7 and it is running on Windows 11.
Have you ever solved this issue?

Of course, in the stable version of 4.4.2 the joystick is recognized.

HTRamsey, 2024-10-29:
As far as I can tell in these cases it is 100% dependent on SDL, and the only thing I would know to try would be different versions of SDL. Without having that joystick it's up to somebody else that actually has one to find a version that works at the moment

hyeonseung89, 2024-12-10:
in file "joystick/cmakelist.txt"

The module settings in the file are as follows. Could you please review them?

Do you have any plans to migrate to sdl3?

```
include(FetchContent)
FetchContent_Declare(SDL
    GIT_REPOSITORY https://github.com/libsdl-org/SDL.git
    GIT_TAG release-2.30.7
    GIT_SHALLOW TRUE
    GIT_PROGRESS TRUE
)
set(SDL2_DISABLE_INSTALL ON CACHE INTERNAL "" FORCE)
set(SDL2_DISABLE_UNINSTALL ON CACHE INTERNAL "" FORCE)
set(SDL2_DISABLE_SDL2MAIN ON CACHE INTERNAL "" FORCE)
set(SDL_SHARED OFF CACHE INTERNAL "" FORCE)
set(SDL_STATIC ON CACHE INTERNAL "" FORCE)
set(SDL_TEST OFF CACHE INTERNAL "" FORCE)

set(SDL_ATOMIC ON CACHE INTERNAL "" FORCE)
set(SDL_AUDIO OFF CACHE INTERNAL "" FORCE)
set(SDL_CPUINFO ON CACHE INTERNAL "" FORCE)
set(SDL_EVENTS ON CACHE INTERNAL "" FORCE)
set(SDL_FILE OFF CACHE INTERNAL "" FORCE)
set(SDL_FILESYSTEM OFF CACHE INTERNAL "" FORCE)
set(SDL_HAPTIC ON CACHE INTERNAL "" FORCE)
set(SDL_HIDAPI ON CACHE INTERNAL "" FORCE)
set(SDL_JOYSTICK ON CACHE INTERNAL "" FORCE)
set(SDL_LOADSO ON CACHE INTERNAL "" FORCE)
set(SDL_LOCALE OFF CACHE INTERNAL "" FORCE)
set(SDL_MISC OFF CACHE INTERNAL "" FORCE)
set(SDL_POWER ON CACHE INTERNAL "" FORCE)
set(SDL_RENDER OFF CACHE INTERNAL "" FORCE)
set(SDL_SENSOR OFF CACHE INTERNAL "" FORCE)
set(SDL_THREADS ON CACHE INTERNAL "" FORCE)
set(SDL_TIMERS OFF CACHE INTERNAL "" FORCE)
set(SDL_VIDEO OFF CACHE INTERNAL "" FORCE)

set(SDL_3DNOW OFF CACHE INTERNAL "" FORCE)
set(SDL_DBUS OFF CACHE INTERNAL "" FORCE)
set(SDL_IBUS OFF CACHE INTERNAL "" FORCE)
set(SDL_MMX OFF CACHE INTERNAL "" FORCE)
set(SDL_VIRTUAL_JOYSTICK ON CACHE INTERNAL "" FORCE)
```

HTRamsey, 2024-12-10:
Migrating to SDL3 is easy, I was just going to at least wait until their stable release

hyeonseung89, 2024-12-12:
The message is shown in

joystickSDL.cc
```
if (SDL_Joystick* sdlJoystick = SDL_JoystickOpen(i)) {
                SDL_ClearError();
                axisCount = SDL_JoystickNumAxes(sdlJoystick);
                buttonCount = SDL_JoystickNumButtons(sdlJoystick);
                hatCount = SDL_JoystickNumHats(sdlJoystick);
                if (axisCount < 0 || buttonCount < 0 || hatCount < 0) {
                    qCWarning(JoystickLog) << "\t libsdl error parsing joystick features:" << SDL_GetError();
                }
                SDL_JoystickClose(sdlJoystick);
            } else {
                **qCWarning(JoystickLog) << "\t libsdl failed opening joystick" << qPrintable(name) << "error:" << SDL_GetError();**
                continue;
            }
```

I tried temporarily changing to sdl3, but I get the same error.

```
if (SDL_Joystick* sdlJoystick = SDL_OpenJoystick(instace_id)) {
                    SDL_ClearError();
                    axisCount = SDL_GetNumJoystickAxes(sdlJoystick);
                    buttonCount = SDL_GetNumJoystickButtons(sdlJoystick);
                    hatCount = SDL_GetNumJoystickHats(sdlJoystick);
                    if (axisCount < 0 || buttonCount < 0 || hatCount < 0) {
                        qCWarning(JoystickLog) << "\t libsdl error parsing joystick features:" << SDL_GetError();
                    }
                    SDL_CloseJoystick(sdlJoystick);
                } else {
                    qCWarning(JoystickLog) << "\t libsdl failed opening joystick" << qPrintable(name) << "error:" << SDL_GetError();
                    continue;
```

```
qgc.joystick.joystickmanager: Joystick added: 0
JoystickLog: Available joysticks
JoystickLog: 	 libsdl failed opening joystick Radiomaster TX16S Joystick error: IDirectInputDevice8::SetCooperativeLevel() DirectX error 0x80070006
JoystickLog: 	 libsdl failed opening joystick RC Simulator - G4.5+G5+G5.5+G6 Controller error: IDirectInputDevice8::SetCooperativeLevel() DirectX error 0x80070006
JoystickLog: 	none found
```

When I connected the simulator joystick and radiomaster tx16s, I got the above debug message.

I don't know the change between sdl2.0 version and sdl2.30 version.

Help me

hyeonseung89, 2024-12-13:
After commenting out all submodule settings in the cmakelist file and compiling, the joystick was recognized. 
As you said above, there seems to be a difference between system provided sdl and custom built, but my skills are not enough to explore this. 
For now, I'm satisfied with the joystick being recognized and I hope you'll make some nice modifications in the future.

HTRamsey, 2024-12-13:
so you're using the system provided SDL & it works properly?

closed 2024-12-13 by HTRamsey

Chasikanaft, 2024-12-16:
> After commenting out all submodule settings in the cmakelist file and compiling, the joystick was recognized. As you said above, there seems to be a difference between system provided sdl and custom [...]

Hi! I have the same problem, which cmakefile did you edit to disable submodules?

Chasikanaft, 2024-12-16:
> @Chasikanaft I'm guessing you can probably temporarily resolve this by removing the minimum version requirement [here](https://github.com/mavlink/qgroundcontrol/blob/b3e66c9e75c5f8f4f51b1be87503ad1483f86dcf/src/Joystick/CMakeLists.txt#L71). Then it should attempt to use your system installed sdl2. I assume the problem might be that one of the required modules of the built SDL2 is disabled but I'd have to be able to recreate the problem first to fix it

```
[cmake] -- Could NOT find SDL2 (missing: SDL2_LIBRARY SDL2_INCLUDE_DIR SDL2_SDLMAIN_LIBRARY) (Required is at least version "2.0.0")
[cmake] -- Checking for module 'sdl2>=2.0.0'
[cmake] --   No package 'sdl2' found
[cmake] -- Building JoystickSDL
```
hm, 4.4.3 "stable" work when installed by installer.exe, i trying to build "master" now, but when i tested continius release installer.exe on another PC i have same error (joystick does not connect anyway)

Chasikanaft, 2024-12-16:
> @Chasikanaft I'm guessing you can probably temporarily resolve this by removing the minimum version requirement [...]

and i have vcpkg installed library
[image]

HTRamsey, 2024-12-16:
it's not set up to use vcpkg