mavlink/qgroundcontrol #11817: Video streaming seems not work in android qt6 qgc

uby56, 2024-08-29:
Hi I trying to build a custom qqc build for android but video streaming is not giving any output 
[image]

And i have followed the instruction for android build [here](https://github.com/mavlink/qgroundcontrol/tree/master/src/VideoManager/VideoReceiver/GStreamer)

Is their any other steps i need for android build

HTRamsey, 2024-08-29:
That happens if gstreamer isn't linked correctly. Can you verify gstreamer is being found by looking at the cmake output?

uby56, 2024-08-30:
Hi @HTRamsey i have check the cmake output where i getting some warning in gstream

Warning: QML import could not be resolved in any of the import paths: org.freedesktop.gstreamer.GLVideoItem
Warning: QML import could not be resolved in any of the import paths: ACME.VideoItem
Warning: QML import could not be resolved in any of the import paths: org.freedesktop.gstreamer.Qt6GLVideoItem

and i check the vedioReciver.pri file and correct the gstream version but it still not working

uby56, 2024-08-30:
yes i am using master branch how can i add the gstreamer pkg currently I am using 1.24.7 or should i download a specific version for the qgc master build

HTRamsey, 2024-08-30:
It should be automatically downloaded during the configure step with cmake. You can look at the CI workflow to double check you are doing it correctly: https://github.com/mavlink/qgroundcontrol/blob/master/.github/workflows/android-linux.yml

uby56, 2024-08-31:
Hi @HTRamsey  I am building the project in my local windows machine. I don't now which file to check is it okay to change the .yml file
[image]

closed 2024-10-16 by HTRamsey