mavlink/qgroundcontrol #11818: Video Source Component doesn't provide different sources.

nechtobolshee, 2024-08-29:
## Expected Behavior
Launching QGC, forwarding a stream in 264 format to port 5600. QGC immediately sees the stream and automatically selects the configuration (understands the format).

## Current Behavior
I created a stream in H264 format and am trying to forward it to QGC. Everything worked on earlier versions, and the video is displayed (automatically when starting a stream on the 5600 port, QGC sees it). Now there is no way to select the H264 video source, and when connecting to Autopilot, the ability to choose a video source is blocked.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Get daily build repository
2. Build 
3. Run .AppImage file
4. Try to pass the stream in (for example) H264 codec. 
5. Open Application Settings -> Video -> Source

## System Information
When posting bug reports, include the following information
- Ubuntu 22.04
- Daily build
- Autopilot (version does not matter)

## Log Files and Screenshots
[image]
[image]

nechtobolshee, 2024-08-29:
@HTRamsey Could you check it, please? 
I saw that you made a large update on this part last month.

HTRamsey, 2024-08-29:
That happens if you don't have GStreamer installed

nechtobolshee, 2024-08-29:
@HTRamsey 
gst-launch-1.0 --version
gst-launch-1.0 version 1.20.3
GStreamer 1.20.3
https://launchpad.net/distros/ubuntu/+source/gstreamer1.0

HTRamsey, 2024-08-29:
Do you have everything in the gstreamer section here installed: https://github.com/mavlink/qgroundcontrol/blob/master/tools/setup/install-dependencies-debian.sh

nechtobolshee, 2024-08-29:
@HTRamsey Yes.. sure.
Ok, could you verify the steps for building the app:

1. git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git
2. git submodule update --recursive
3. sudo ./tools/setup/install-dependencies-debian.sh
4. cmake -B build -G Ninja -DCMAKE_BUILD_TYPE=Debug
5. cmake --build build --config Debug
6. ./build/QGroundControl

Right?

nechtobolshee, 2024-08-29:
Uploading 2024-08-29.mp4…

Did u see something like that?

nechtobolshee, 2024-08-29:
https://github.com/user-attachments/assets/d339f10e-7a09-4332-b612-47368ae31db1

Have you encountered something like that? I have tried to unpack stream by gst-launch-1.0 - all is ok.

HTRamsey, 2024-08-29:
Close, you can look here for an example: https://github.com/mavlink/qgroundcontrol/blob/master/.github/workflows/linux.yml.

closed 2024-08-30 by nechtobolshee