mavlink/qgroundcontrol #11825: I can't add a fence in the Daily build, but it works correctly in v428

Drone-Lab, 2024-08-30:
## Expected Behavior
When I connect QGC with PX4 jmavsim and  try to add a fence on QGC, I find the buttonis gray and I can't click it.

[image]

HTRamsey, 2024-09-02:
I don't personally have this problem nor do I see any obvious changes related to it from the daily but maybe @DonLakeFlyer has an idea

DonLakeFlyer, 2024-10-13:
This works fine for me with MockLink. I can't test jmavsim on OSX. The code is as follows:
```
bool GeoFenceController::supported(void) const
{
    return (_managerVehicle->capabilityBits() & MAV_PROTOCOL_CAPABILITY_MISSION_FENCE) && (_managerVehicle->maxProtoVersion() >= 200);
}
```

My guess is that jmavsim is not running mavlink v2.

closed 2024-11-30 by HTRamsey