mavlink/qgroundcontrol #11830: Outdated QGC app version on google playstore

KarthiAru, 2024-09-02:
Hi,

I noticed that the version of the QGC app available on Google Play Store is v4.1.1, which was last updated in January 2021. Could you please share why there haven't been any updates for this app over the past few years?
[image]

kschoelpple, 2024-09-03:
Can't reply officially for the QGC project, but here is what I know:
Google keeps updating their requirements and as QGC is an open source project with only a few developers working on it, it has been falling behind in updating certain elements to suit those requirements. --> Hence QGC for Android is available from an alternative download site.
My understanding is that this year has seen an increase in activity on this project and hopefully in the near future it will be once again in line with Google's ever changing requirements.

HTRamsey, 2024-09-03:
I want to say that when I looked into this I found that the guy that owns the google account that lists QGC is no longer active, but I'll see if we can get it figured out for the next big stable release. Google's requirements aren't anywhere near as bad as Apple's, so I think listing on the play store is feasible but almost definitely not on the App Store for iOS.

DonLakeFlyer, 2024-09-03:
What @kschoelpple says is exactly correct. Hopefully we can get it working aging with this next release.

HTRamsey, 2024-10-09:
@KarthiAru I just asked about this and it's because dronecode needs to make a new google account for the playstore. Then I can immediately start uploading the builds again

DonLakeFlyer, 2024-10-09:
@dagar You're setup as the administrator for the QGC google play account. I tried getting back into it a few weeks back but couldn't. I think my access may have timed out or something. Do you think you could both myself and Holden as admins in there so QGC can be fixed up to get back into the store?

dagar, 2024-10-11:
> @dagar You're setup as the administrator for the QGC google play account. I tried getting back into it a few weeks back but couldn't. I think my access may have timed out or something. Do you think [...]

Can you check again?

DonLakeFlyer, 2024-10-12:
Thanks @dagar I'm back in again.

jidanni, 2026-09-19:
Can you people please get your act together.

You wrote all that code but then can't manage to put it (back) in the Play Store.

One might have thought there might have been some ideological reason for not putting it there.

But it turns out it is simple confusion.

DonLakeFlyer, 2026-09-19:
@jidanni We will gladly refund your purchase price for QGroundControl if you are unsatisfied.