mavlink/qgroundcontrol #11832: Open failed on port ttyACM0: Permission denied

eshcheglov, 2024-09-04:
Hi. 

I can't flash PX4 Autopilot to my Pixhawk using QGC

## Expected Behavior
PixHawk flashed with PX4 Autopilot

## Current Behavior
[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open QGC
2. Go to Vehicle Setup -> Firmware
3. Connect Pixhawk via USB

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 24.10
- QGC Version: 4.4.0 (AppImage)
- QGC build: stable
- Flight Controller: Pixhawk 6c mini

## Detailed Description
- User is in the `dialout` group
- Permissions for /dev/ttyACM0 is correct: 
```
$ ls -la /dev/ttyACM*
crw-rw---- 1 root dialout 166, 0 вер  4 12:44 /dev/ttyACM0
crw-rw---- 1 root dialout 166, 1 вер  4 12:44 /dev/ttyACM1
```
- modemmanager is absent 
- I can connect to `/dev/ttyACM0` and `/dev/ttyACM1` via minicom (so I can assume that no-one else is using those serials)

## Log Files and Screenshots
<details>
  <summary>sudo journalctl -f</summary>

``` 
12:49:28 carbon kernel: usb 1-2: new full-speed USB device number 54 using xhci_hcd
12:49:28 carbon kernel: usb 1-2: New USB device found, idVendor=3185, idProduct=0038, bcdDevice= 1.01
12:49:28 carbon kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
12:49:28 carbon kernel: usb 1-2: Product: PX4 BL FMU v6C.x
12:49:28 carbon kernel: usb 1-2: Manufacturer: Auterion
12:49:28 carbon kernel: usb 1-2: SerialNumber: 0
12:49:28 carbon kernel: cdc_acm 1-2:1.0: ttyACM0: USB ACM device
12:49:28 carbon mtp-probe[35955]: checking bus 1, device 54: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
12:49:28 carbon mtp-probe[35955]: bus: 1, device: 54 was not an MTP device
12:49:28 carbon mtp-probe[36015]: checking bus 1, device 54: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
12:49:28 carbon mtp-probe[36015]: bus: 1, device: 54 was not an MTP device
12:49:33 carbon kernel: usb 1-2: USB disconnect, device number 54
12:49:33 carbon kernel: usb 1-2: new full-speed USB device number 55 using xhci_hcd
12:49:33 carbon kernel: usb 1-2: New USB device found, idVendor=3162, idProduct=0053, bcdDevice= 2.00
12:49:33 carbon kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
12:49:33 carbon kernel: usb 1-2: Product: Pixhawk6C-bdshot
12:49:33 carbon kernel: usb 1-2: Manufacturer: Holybro
12:49:33 carbon kernel: usb 1-2: SerialNumber: 3A0035001751323531393234
12:49:33 carbon kernel: cdc_acm 1-2:1.0: ttyACM0: USB ACM device
12:49:33 carbon kernel: cdc_acm 1-2:1.2: ttyACM1: USB ACM device
12:49:33 carbon mtp-probe[36058]: checking bus 1, device 55: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
12:49:33 carbon mtp-probe[36058]: bus: 1, device: 55 was not an MTP device
12:49:33 carbon mtp-probe[36139]: checking bus 1, device 55: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
12:49:33 carbon mtp-probe[36139]: bus: 1, device: 55 was not an MTP device
```

</details>

<details>
  <summary>QGC log</summary>

```
$ QGroundControl.AppImage
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Settings location "/home/copy/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/copy/Documents/QGroundControl/Logs"
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
Ignoring ssl certificates due to OpenSSL version mismatch
Map Cache in: "/home/copy/.cache/QGCMapCache300" / "qgcMapCache.db"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
qt.network.ssl: QSslSocket: cannot call unresolved function EVP_PKEY_base_id
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error 0
"v4.3.0"
Ignoring ssl certificates due to OpenSSL version mismatch
Ignoring ssl certificates due to OpenSSL version mismatch
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
qt.network.ssl: QSslSocket: cannot call unresolved function EVP_PKEY_base_id
FirmwareUpgradeLog: PX4 releases github download failed "Error during download. Error: 201"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
qt.network.ssl: QSslSocket: cannot call unresolved function EVP_PKEY_base_id
"Holybro" "Pixhawk6C-bdshot"
QIODevice::write (QSerialPort): device not open
"Write failed: Permission denied"
bool QSerialPort::flush(): device not open
VideoReceiverLog: Stop called on empty URI
```

</details>

DonLakeFlyer, 2024-09-06:
This is almost always some sort of OS permission problem. Can you try with a daily build instead?

eshcheglov, 2024-09-27:
I tried today daily build. Still same

eshcheglov, 2024-10-17:
Okay, after the latest Ubuntu 24.10 update, the problem disappeared

journalctl now looks different: 

```
Oct 17 12:01:49 carbon kernel: usb 1-4: new full-speed USB device number 25 using xhci_hcd
Oct 17 12:01:50 carbon kernel: usb 1-4: New USB device found, idVendor=3185, idProduct=0038, bcdDevice= 1.01
Oct 17 12:01:50 carbon kernel: usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 17 12:01:50 carbon kernel: usb 1-4: Product: PX4 BL FMU v6C.x
Oct 17 12:01:50 carbon kernel: usb 1-4: Manufacturer: Auterion
Oct 17 12:01:50 carbon kernel: usb 1-4: SerialNumber: 0
Oct 17 12:01:50 carbon kernel: cdc_acm 1-4:1.0: ttyACM5: USB ACM device
Oct 17 12:01:50 carbon mtp-probe[279336]: checking bus 1, device 25: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4"
Oct 17 12:01:50 carbon mtp-probe[279336]: bus: 1, device: 25 was not an MTP device
Oct 17 12:01:50 carbon mtp-probe[279382]: checking bus 1, device 25: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4"
Oct 17 12:01:50 carbon mtp-probe[279382]: bus: 1, device: 25 was not an MTP device
Oct 17 12:01:50 carbon Thunar[279378]: thunar-volman: Unsupported USB device type "usb".
Oct 17 12:01:51 carbon Thunar[279394]: thunar-volman: Unsupported USB device type "cdc_acm".
Oct 17 12:01:51 carbon Thunar[279400]: thunar-volman: Unsupported USB device type "cdc_acm".

```

Line ` USB disconnect` is no longer present. I suppose that something changed in `cdc_acm` or in udev rules

closed 2024-10-17 by eshcheglov