mavlink/qgroundcontrol #11839: Changelog out of date

Pandapip1, 2024-09-05:
Hi!

I'm updating the packaging for QGC in nixpkgs, and noticed that there was a changelog I could add:

https://github.com/mavlink/qgroundcontrol/blob/7baf97e262e1534144a1caea8e27d9174e6596a2/ChangeLog.md?plain=1#L1-L204

However, this changelog appears to be out of date. Should it be removed to avoid confusion?

DonLakeFlyer, 2024-09-05:
It's not out of date. It's just poorly maintained with respect to Stable changes. It's shouldn't be removed

Pandapip1, 2024-09-06:
I would expect a changelog to at least list all releases with a nominal "fixed bugs" or something.

DonLakeFlyer, 2024-09-06:
Happy to take a pull with that, although I don't think it adds much value. As I said "poorly maintained" but better than nothing.

closed 2025-07-09 by HTRamsey