mavlink/qgroundcontrol #11857: Release v4.4.2 video stream is black (doesn't show) in detached window

kwh44, 2024-09-08:
----

## Expected Behavior
I launched QGC, video stream is shown in small embedded window, but when clicking on detaching video stream into separate window, video stream is expected to show in detached window as in embedded (small resizable screen on top of map)

## Current Behavior
Please describe what happens instead of the expected behavior.
I launched QGC, video stream is shown in small embedded window, but when clicking on detaching video stream into separate window, video stream is black, doesn't show in that window

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Launch QGC
2. Wait for video stream to show in small embedded window
3. Click on button to detach video stream into separate window
4. Now you can see that video stream is black in detached window, while in small embedded window it was showing the stream.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 24.04.1 LTS
- QGC Version: 4.4.2 official build
- QGC build: downloaded from https://github.com/mavlink/qgroundcontrol/releases/download/v4.4.2/QGroundControl.AppImage
- Flight Controller: SITL
- Autopilot (with version):  SITL ardupilot

## Detailed Description
video stream doesn't show in detached window, I am using QT 5.15.2, gstreamer version is  1.24.2

## Log Files and Screenshots
Video stream is shown in embedded window
[image]
Video stream is not shown (black) in detached window
[image]

kwh44, 2024-09-08:
https://github.com/mavlink/qgroundcontrol/issues/10803 looks to be previously reported

aXynt, 2025-03-30:
I also have the same problem on ubuntu 22.04 and 24.04 with v4.4 or daily version. @HTRamsey do you have any solution?

HTRamsey, 2025-03-30:
Yeah it was one of the things I was going to fix before v5. Probably will focus on mostly video in general for a while

aXynt, 2025-05-18:
@HTRamsey 5.0 stable is coming soon, will you fix this for 5.0?