mavlink/qgroundcontrol #11864: Takeoff Altitude cannot be lower than 10m

twistedwings, 2024-09-10:
When triggering a takeoff command a slider appears to select the takeoff altitude.

## Expected Behavior
Selectable range should be \<Minimum Altitude\> to \<Maximum Altitude\> from Application Settings-Guided Command Settings

## Current Behavior
Selectable range is 10 to 120m.

## Steps to Reproduce:
Trigger takeoff command

## System Information
Windows 10
- QGC Version: 4.4.2
- QGC build: stable
- Flight Controller: SITL
- Autopilot (with version): Ardupilot Copter 4.5.6

## Log Files and Screenshots
[image]

[image]

DonLakeFlyer, 2024-09-13:
There were a bunch of bugs here related to this.You should try a daily build to see if this has already been fixed.

twistedwings, 2024-09-13:
yes, it's fixed in the daily.

closed 2024-09-17 by DonLakeFlyer

federicociresola, 2024-10-04:
When will the fix of this bunch of bugs be released in a new stable version?