mavlink/qgroundcontrol #11865: Error in longitude data in georreferenced images

jorgitinlugo, 2024-09-11:
Hi, I have a problem with my images, my gps captures this location:

[image]

The problem is that when i make a photo and see the gps metadata parameters i see this:

[image]

Latitude is correct but the Longitude info doesnt have sense.

Should i check my camera or its a qground failure?

HTRamsey, 2024-09-11:
Version?

jorgitinlugo, 2024-09-11:
Im using the Siyi camera ZT6 with the latest version and 4.4.1 Qground versión, in longitude the real position is -8.4 W, in the image params seems to be 4 times smaller.

closed 2026-10-05 by github-actions[bot]