mavlink/qgroundcontrol #11871: QGC master branch build for Android MavLink not seeing in Analyze page

Ehsan-Home, 2024-09-12:
Hi everyone,

First of all, thank you for this amazing library! I'm not sure if this is the right place to ask questions about building QGroundControl (QGC), so please let me know if there's a better forum for this topic.

I recently built the QGC master branch from source [Link](https://github.com/mavlink/qgroundcontrol) using the Qt IDE on my Android tablet. I have a HereLink controller connected to the tablet, as shown in the diagram below:

Tablet <===========> HereLink controller

The issue I'm facing is with the MavLink connection establishment in QGC. When I use my custom build, it does not display the heartbeat of the HereLink controller, as shown in the screenshot below:

[image]

However, when I install the official QGC APK from their website ([Download and Install | QGC Guide (master)](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/getting_started/download_and_install.html)), the "Analyze" page starts showing the heartbeat from the controller without any issues:

[image]

Here's where things get even more confusing: If I first install the official QGC APK and confirm that the heartbeat is working, then uninstall it and install my custom-built version (from the master branch), the heartbeat still shows up correctly on the "Analyze" page! This makes it seem like there's an issue with initializing or detecting the controller heartbeat when using my custom build from the start.

Am I doing something wrong in the building?
I will provide any logs needed.
Thank you all.

DonLakeFlyer, 2024-09-13:
> I have a HereLink controller connected to the tablet, as shown in the diagram below:

Connected how? Are you using the automatically forwarded UDP connection from the Herelink to the tablet? No idea why you would get these sorts of results.

Ehsan-Home, 2024-09-25:
Hi. Thanks for the reply.

The tablet and Controller are connected using ethernet cable. 

> Are you using the automatically forwarded UDP connection from the Herelink to the tablet?
Not sure about this. Can you elaborate it please?

Also, please let me know what other information you need so I can provide. 
Thanks

DonLakeFlyer, 2024-10-09:
Ok, got it. Really hard to say what could be going wrong. You are going to need to debug though it somehow. Can't really help much on custom builds since they differ from regular QGC. First thing to do would be to build QGC master APK yourself to see if that works for you. If not it indicates some problem in your build procedure. Other than that not much help I can provide.

closed 2024-10-09 by DonLakeFlyer