mavlink/qgroundcontrol #11873: battery status not refreshed

twistedwings, 2024-09-13:
## Expected Behavior
battery status refreshed regularly

## Current Behavior
battery status refreshed only sporadically

## Steps to Reproduce:
Compare battery status with MAVLink message

## System Information
- Operating System: Windows 10
- QGC Version: 4.4.2
- QGC build: stable
- Flight Controller: Matek F405-CTR
- Autopilot (with version): Ardupilot Copter 4.5.6

## Log Files and Screenshots
[image]

Log File
[2024-09-13 13-42-03.zip](https://github.com/user-attachments/files/16995940/2024-09-13.13-42-03.zip)

DonLakeFlyer, 2024-09-13:
Battery status comes from the BATTERY_STATUS message not SYS_STATUS. Running into problems replaying blog. Looking at this right now. Can you double-check BATTERY_STATUS is up to date?

DonLakeFlyer, 2024-09-13:
I replayed the log although SYS_STATUS streams the values BATTERY_STATUS seems to stop streaming values to QGC at some point. So this is a vehicle side problem, not QGC as far as I can tell.

twistedwings, 2024-09-13:
yes, you are right. The BATTERY_STATUS stopped streaming for whatever reason.

DonLakeFlyer, 2024-09-17:
Closing since not much QGC can do if telemetry is not up to date. Re-open is there is some proof that this is a QGC side problem.

closed 2024-09-17 by DonLakeFlyer