mavlink/qgroundcontrol #11882: QGC is unable to find SYS_HAS_NUM_ASPD parameter on PX4 (1.14.3)

trileis, 2024-09-15:
## Expected Behavior
No error displayed after firmware upgrade

## Current Behavior
A `SYS_HAS_NUM_ASPD` missing parameter is thrown (see screenshot in reproducer)

## Steps to Reproduce:
With "Fixed Wing" airbone,
1. Flash FC with latest firmware version (1.14.3): 
[image]

2. Go back to summary and get a popup with the error: 
[image]

## System Information
- Operating System: Windows10
- QGC Version: 4.4.2/4.4.1
- QGC build: stable
- Flight Controller: Pixhawk 6X
- Autopilot (with version): PX4 1.14.3

## Detailed Description
Apparently the parameter SYS_HAS_NUM_ASPD is not present on the PX4 version 1.14.3, it should be merged in 1.15.0. In QGC code diff between version 4.4.0 -> 4.4.1 ([here](https://github.com/mavlink/qgroundcontrol/compare/v4.4.0...v4.4.1#diff-9c83171468cf135766163fce20c29d00cb6f45f6ed4b90459c311a40919f4c25)), this parameter is searched when PX4 version is upper or equal to 1.14. It should better be 1.15.

DonLakeFlyer, 2024-09-15:
Fixed in daily builds

trileis, 2024-09-16:
I don't see any modification on that part of code ((PX4/SensorComponents.cc)[src/AutoPilotPlugins/PX4/SensorsComponent.cc]) on master since commit: da57ee. I think the problem still there on daily build.

DonLakeFlyer, 2024-09-16:
> I think the problem still there on daily build.

And you actually ran the daily build to verify that?

DonLakeFlyer, 2024-09-16:
https://github.com/mavlink/qgroundcontrol/pull/11425

trileis, 2024-09-16:
Yes, I ran it two days before with the last master release. I think while PX4 doesn't release (at the moment) the parameter `SYS_HAS_NUM_ASPD`, this couldn't work. The `SYS_HAS_NUM_ASPD`  parameter will be part of PX4 v1.15.0, which is at the moment in beta test.

To fix it, I downgrade QGC to 4.4.0 and it works like a charm with PX4 v1.4.3.

I'm supose the [#11425](https://github.com/mavlink/qgroundcontrol/pull/11425) has been release too late. A possible fix could be to modify src/AutoPilotPlugins/PX4/SensorsComponent.cc to replace the check on PX4 version v1.14 by a check on v1.15.

I would have the time to do that as a dev but I'm trying to make my plane flying. I'm just playing with the Pixhawk system since a week so I'm not a dev at that time.

DonLakeFlyer, 2024-09-16:
> there's possibly a bug

Yup, I'll fix it

DonLakeFlyer, 2024-09-16:
Can you pull the build artifact from this pull once it's done and verify that it works for you?

trileis, 2024-09-16:
Nice, I'll test that.

trileis, 2024-09-17:
I tested the artifact from pull request (v4.3.0-889-gece5b6e55) and the problem is fixed. Thanks.

DonLakeFlyer, 2024-09-17:
Great, thanks so much for testing. Sorry for screwing this up.

PR #11885 (2024-09-17): Airspeed cal: Fix version check
Fix #11882
closed 2024-09-17 by DonLakeFlyer