mavlink/qgroundcontrol #11897: CRC Validation of MetadataTypes Does Not Obey Schema

RbtsEvrwhr-Riley, 2024-09-18:
We have encountered an issue where the fileCrc field in the metadata_general schema is required in QGC but not required by the MavLink protocol reference. We have isolated the exact point of failure, and have worked around it locally for now. What do we want to do about this issue?

## Expected Behavior
A metadata_general.json file that follows the schema correctly should be parsed and loaded.

## Current Behavior
A metadata_general.json file that does not contain the **optional** field of fileCrc for every file in metadataTypes will not be parsed. Crcvalid is checked in CompInfoGeneral.cc CompInfoGeneral::setJson line 73

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Configure any UAS with a general metadata file like the attached development file.
2. Connect to a new instance of qGroundControl such that it the file downloads using ftp or loads from cache
3. The parameter and actuator files will not be loaded.

## System Information
When posting bug reports, include the following information
- QGC all versions since 9/12/2022 Daily
- Windows and Linux
- Any flight controller that supports COMPONENT_METADATA

## Detailed Description
https://github.com/mavlink/mavlink/blob/master/component_metadata/general.schema.json here is the reference schema.

## Log Files and Screenshots
- We have moved past this issue in development, if you need logs regenerated we can do that but repro steps and the root cause have already been isolated.

[metadata_general.json](https://github.com/user-attachments/files/17046832/metadata_general.json)

DonLakeFlyer, 2024-09-18:
@bkueng Can you look at this?

bkueng, 2024-09-20:
You are correct, QGC should not reject it if the CRC is missing. Do you want to create a PR?

Please also be aware the a missing CRC is generally intended for dynamically changing metadata, which is not implemented in MAVLink. A missing CRC at this point means QGC will have to download the file each time it connects to the vehicle.

DonLakeFlyer, 2024-09-20:
> Please also be aware the a missing CRC is generally intended for dynamically changing metadata, which is not implemented in MAVLink. A missing CRC at this point means QGC will have to download the file each time it connects to the vehicle.

`"description":  "CRC32 checksum of the file. This should not be set for dynamically updated metadata."`

The json schema description is not a clear as this statement. Should that be updated with better wording like "dynamically changing" and no GCS caching?

bkueng, 2024-09-27:
Yes I can do that

closed 2026-10-05 by github-actions[bot]