mavlink/qgroundcontrol #11942: Unable to install in recent version of Android

antonymarion, 2024-09-23:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/support/support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Android release downloaded from github does not install.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Go To release 
2. Download the apk
3. Select the apk on android 
4. Warning message and app cannot install

## System Information
When posting bug reports, include the following information
- Operating System: Android 14

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
[image]

HTRamsey, 2024-09-24:
I don't know what that says

antonymarion, 2024-09-24:
It says "This application has been built for an older version of Android and does not include last privacy protection standards"

HTRamsey, 2024-09-24:
@julianoes you know how we delete all SDK's above 33 for the release? We need to not do that for Android 14

LuizaPac, 2024-10-10:
Is there a list of supported android systems? I'm facing the same problem here, but if there is no expectation to solve it, I may be able to aquire a diferent  android to run it.

HTRamsey, 2024-10-10:
@LuizaPac there's just a minimum of something like Android 7 or somewhere along those lines. It's like a 5 second fix, just need to remove these lines from the release: https://github.com/mavlink/qgroundcontrol/blob/990ed12f5be9464d737ba0da703a126f63b0eff0/.github/workflows/android_release.yml#L77

dimitri-justeau, 2024-12-06:
Hi, was this issue solved ? I am running into the same problem with an Android 12 device.

antonymarion, 2024-12-06:
cf 

> @LuizaPac there's just a minimum of something like Android 7 or somewhere along those lines. It's like a 5 second fix, just need to remove these lines from the release:
> 
> https://github.com/mavlink/qgroundcontrol/blob/990ed12f5be9464d737ba0da703a126f63b0eff0/.github/workflows/android_release.yml#L77

dimitri-justeau, 2024-12-06:
Thank you, I've seen this answer but I'm not sure whether we should we expect that it will be done on this repo, or if should we open a PR, or if we should fork the repo to do it outside of the official repo ?

dimitri-justeau, 2024-12-06:
Please forget my post, actually it's not very intuitive but it is possible to click on the "install anyway" text, just above the Ok button in Android, and it does install the app !

closed 2025-04-22 by HTRamsey