mavlink/qgroundcontrol #11944: [ Android ] QGC Occurring common android source build error，but still having problem solving it ，despite with online raised solutions

dannynfs, 2024-09-24:
Occurring common android source build error，but still having problem solving it ，despite with online raised solutions
Im running qgc stable 4.4/qt 5.15.0,android build and having following error
[image]
the following is my android env settings
[image]
[image]
By researching online i found that it usually happened because of gradle version incompatible with jdk version ，so two common method solving it is by upgrading the projects gradle version or degrade java jdk into jdk11，where i came across issues on attempting both methods

1.Upgrading the projects gradle version:
By modifying both "gradle-wrapper.properties" and "build.gradle"s gradle specification 
[image]
[image]
but after running the project，the changes i made both went back to the original settings which is 5.6.4 in gradle wrapper.properties and 3.6.0 in the latter，don't know if there's any other file included in qgc stable 4.4 does that

2.degrade java jdk into jdk11
By attempting it this occurs
[image]
[image]
[image]
simply cant degrade to jdk11 

both issues have been searching online but no obvious solution found 
maybe some mighty heros could help me here

HTRamsey, 2024-09-24:
#11907

closed 2024-09-24 by HTRamsey

aibietdau, 2024-11-13:
> #11907
i tried but the result is not as i expected, if i use another jdk for v4.4 is it ok