mavlink/qgroundcontrol #11949: Run fails on Android QT

antonymarion, 2024-09-24:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/support/support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Android Build over QT and windows should run

## Current Behavior
Build (compilation) is OK but when launched it fails

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open QT
2. Compile for android
3. Error logs on Qt

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/settings_view/console_logging.html)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

antonymarion, 2024-09-24:
[image]

HTRamsey, 2024-09-24:
Version?

antonymarion, 2024-09-25:
Last version : 4.4.2

HTRamsey, 2024-09-26:
4.4.2 doesn't use Qt6 and also is better with qmake than cmake

antonymarion, 2024-09-26:
Well thks for the reply, to save time I will use github actions CI where build are configured with yaml ;)

closed 2024-09-26 by antonymarion