mavlink/qgroundcontrol #11951: Config error: Failed to find required Qt component "MultimediaQuickPrivate"

liukai12345614, 2024-09-25:
## Expected Behavior
QGC can be successfully built using the cmake command

## Current Behavior
Error after building with cmake:
-- QGC_ENABLE_QT_VIDEOSTREAMING="ON"
-- Could NOT find Qt6MultimediaQuickPrivate (missing: Qt6MultimediaQuickPrivate_DIR)
CMake Error at C:/gitP/Microsoft/vcpkg/scripts/buildsystems/vcpkg.cmake:859 (_find_package):
  Found package configuration file:

    C:/gitP/Microsoft/vcpkg/installed/x64-windows/share/Qt6/Qt6Config.cmake

  but it set Qt6_FOUND to FALSE so package "Qt6" is considered to be NOT
  FOUND.  Reason given by package:

  Failed to find required Qt component "MultimediaQuickPrivate".

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.git -C "C:\gitP" clone --no-checkout https://github.com/mavlink/qgroundcontrol mavlink\qgroundcontrol
2.git -C "C:\gitP\mavlink\qgroundcontrol" checkout 3139615
3.git -C "C:\gitP" clone https://github.com/Microsoft/vcpkg Microsoft\vcpkg 
4.set VSCMD_SKIP_SENDTELEMETRY=1 & "C:\Program Files\Microsoft Visual Studio\2022\Enterprise\Common7\Tools\VsDevCmd.bat" -host_arch=amd64 -arch=amd64 & set _CL_= /Bcapture_repro C:\a\_work\_temp\rwc_project_logs\Qgroundcontrol\preprocessed_repro_setup & set _LINK_= /onfailrepro:C:\a\_work\_temp\rwc_project_logs\Qgroundcontrol\link_repro_setup
5.set PATH=C:\Program Files\Microsoft Visual Studio\2022\Enterprise\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;%PATH%
6.cd /d C:\gitP\Microsoft\vcpkg
7.bootstrap-vcpkg.bat 2>&1
8.vcpkg.exe install --x-asset-sources=x-azurl,https://vcpkg.storage.devpackages.microsoft.io/artifacts/;x-block-origin --x-buildtrees-root=C:\vcpkgbt --keep-going --recurse qtcharts qtlocation qtpositioning qtspeech qt5compat qtmultimedia qtserialport qtimageformats qtshadertools qtconnectivity qtquick3d qtsvg qtsensors --triplet x64-windows --clean-after-build 2>&1
9.cd /d C:\gitP\mavlink\qgroundcontrol\build_amd64
10.cmake -G "Visual Studio 17 2022" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.22621.0 -DCMAKE_TOOLCHAIN_FILE=C:\gitP\Microsoft\vcpkg\scripts\buildsystems\vcpkg.cmake -DVCPKG_TARGET_TRIPLET=x64-windows -DCMAKE_BUILD_TYPE=release -DQGC_STABLE_BUILD=OFF .. 2>&1

## System Information
When posting bug reports, include the following information
- Operating System: Windows Server 2022 Datacenter
- QGC Version: 3139615
- QGC build: stable

## Detailed Description
From the execution log, MultimediaQuickPrivate is required to configure QGC. Should I install a certain library using vcpkg? Or what should I do? Could you please help take a look this issue? Thanks in advance!

## Log Files and Screenshots
[vcpkg.log](https://github.com/user-attachments/files/17123633/vcpkg.log)
[Setup.log](https://github.com/user-attachments/files/17123635/Setup.log)

HTRamsey, 2024-09-25:
For now just disable QGC_ENABLE_QT_VIDEOSTREAMING and use gstreamer instead, it's not ready anyways. But yeah we noticed a few systems don't have MultimediaQuickPrivate for whatever reason and won't support that method of video streaming for now

liukai12345614, 2024-10-16:
Hi @HTRamsey , thanks for you info. I try to disable QGC_ENABLE_QT_VIDEOSTREAMING and use gstreamer instead to config it. The result was successful. Close this issue.

closed 2024-10-16 by liukai12345614