mavlink/qgroundcontrol #11956: Gimbal manual control doesn't work

eshcheglov, 2024-09-27:
## Steps to Reproduce:

Follow the [official instruction](https://docs.px4.io/main/en/advanced/gimbal_control.html)
1. Set `MNT_MODE_OUT=AUX`
2. Set `MNT_MODE_IN=Auto`
3. Configure Gimbal Servo using `Actuators` tab:
```
PWM:       200Hz
AUX 5:     "Gimbal Pitch"
Disarmed:  1000
Minimum:   1100
Maximum:   1900
```
4. Verify that gimbal servo is working using `Actuator Testing`
5. Set `MNT_MAN_PITCH=AUX5`
6. In `Joystick -> Button Assigment` tab set `Gimball Up` and `Gimball Down` to some buttons on connected USB joystick 

Gimbal work automatically:
- It change position after arming
- It changing position relatively to frame position if `MNT_DO_STAB` is enabled. 
So I can assume that gimbal is actually working, but I can't manipulate it manually for some reason

## System Information
- Operating System: Ubuntu 24.10
- QGC Version: 4.4.0 (AppImage)
- QGC build: stable
- Flight Controller: Pixhawk 6c mini
- Flight Controller FW: PX4 Autopilot 1.14.3

julianoes, 2024-09-30:
Can you try with PX4 v1.15.0 and/or with MNT_MODE_IN set to MAVLink gimbal v2?

eshcheglov, 2024-10-01:
> but there's no response to any of the options.

Alright, after arming, the gimbal responds to the `Center` and `Tilt 90` buttons, correctly adjusting its position. However, there's still no response to the joystick buttons

julianoes, 2024-10-01:
Aha, that's good news! Which gimbal is that? It sounds like angle input works but angular velocity does not. I wonder if the problem is on the PX4 side converting this.

I remember looking into this a while ago. Could you double check with PX4 v1.15.0?

julianoes, 2024-10-17:
Ok, I believe the issue here is that your gimbal output (PWM output) doesn't support angular rate commands but only angles. So we should probably add that to the PX4 driver.

eshcheglov, 2024-10-17:
Hi!

> Aha, that's good news! Which gimbal is that? 

It's a custom build based on a standard PWM Servo called `PDI-1181MG` produced by "JX-SERVO"
It has the following parameters: 
```
Dead band: 2μs
Working frequence: 1520μs / 300hz
Motor: core motor
Operating Voltage: DC4.8~6.0 V
Operating Speed (4.8V): 0.13sec/60°
Operating Speed (6V): 0.11 sec/60°
Stall Torque (4.8V): 3.0kg.cm
Stall Torque (6V): 3.5 kg.cm 
```

Servo rail is powered by 5V BEC

> I remember looking into this a while ago. Could you double check with PX4 v1.15.0?

Yeah, all current test results is from v1.15.0 

> Ok, I believe the issue here is that your gimbal output (PWM output) doesn't support angular rate commands but only angles. 

I achieved some minimal (almost invisible, but still) response to joystick input after changing `MNT_RANGE_PITCH` from default `90` to `20` deg. However, delay between button press and reaction is so long, and servo angle change is so small, that I need to press "down" button at least 15 times with 2 sec delay between every press to get 20 deg on gimbal. Unfortunately, I still can't understand what this parameter actually does..

MaEtUgR, 2024-10-21:
> In Joystick -> Button Assigment tab set Gimball Up and Gimball Down to some buttons on connected USB joystick

FYI in my testing with a MAVLink gimbal protocol v2 output the gimbal up/down commands from the joystick only move the gimbal by 1°. I'd need to check what the implementation on ground station side is bug I thought it takes the current gimbal attitude as per it's telemetry and just adds a tiny amount on top and you have to actually keep on spamming the button on repeat to see the movement. To be verified but that's an observation I made. The "gimbal center" joystick button command sends an attitude command to level out and executes nicely [if no conflicting commands are coming simultaneously](https://github.com/PX4/PX4-Autopilot/pull/23825).

julianoes, 2024-10-29:
@eshcheglov I can now reproduce this. When you check the "Repeat" checkbox next to the gimbal commands it starts working, however, it's still painfully slow.

@Davidsastresas should we just increase the step size to make this faster somehow?

MaEtUgR, 2024-10-30:
> I wonder if we should instead use the angular velocity API for this, so when you press it down it will start the movement and when you release the button it will stop the movement.

Sounds like a better approach yes. I think for a good user experience, you need to be able to quickly bring it up to level, straight down and possibly also 45° forward down and then fine adjust with whatever button or rocker you have. The speed for the adjustment needs to be tied to focal lengt/zoom level.

eshcheglov, 2024-10-31:
BTW, the on-screen gimbal control works like a charm. I'm referring to the touchscreen gesture approach, where touching the top or bottom of the screen moves the gimbal quickly and across a wide range.

julianoes, 2024-11-04:
@eshcheglov hm, can you maybe provide a video regarding the touchscreen gesture problem? I'm not sure I understand what you mean.

closed 2024-11-06 by julianoes