mavlink/qgroundcontrol #11963: Upload ROI failed

twistedwings, 2024-10-01:
Uploading of ROI failed.
Looks similar to
](https://github.com/mavlink/qgroundcontrol/issues/11159)----

## Expected Behavior
Upload mission without error

## Current Behavior
Upload mission with alert

## Steps to Reproduce:
1. create mission with ROI
2. upload mission

## System Information
When posting bug reports, include the following information
- Windows 10
- QGC Version:
[image]
- daily
- SITL
- ArduCopter 4.6.0

## Log Files and Screenshots
[image]

DonLakeFlyer, 2024-10-29:
@bytesByHarsh Can you see if you can repro this with daily and ArduPilot SITL? If you can, ping me on discord and we can discuss where to start looking.

DonLakeFlyer, 2024-10-30:
Looking through the code with respect to ArduPilot not supporting DO_SET_ROI_LOCATION, DO_SET_ROI_NONE and MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET this means that StructureScan can't generate a mission which is supported by ArduPilot. It uses those to point at the structure correctly.

DonLakeFlyer, 2025-03-11:
@twistedwings Can you verify this?

twistedwings, 2025-03-11:
Test in SITL with ArduCopter 4.6.0-beta4 worked without errors.

DonLakeFlyer, 2025-03-11:
Thanks so much for reporting and helping with testing

closed 2025-03-11 by DonLakeFlyer

brad112358, 2025-03-11:
Can we get an official QGC release with this problem fixed?

DonLakeFlyer, 2025-03-12:
We're in the process of wrapping up the 5.0 release as it stands now. Hoping to have that out in May.