mavlink/qgroundcontrol #11965: Can I process MAVLINK Messages using Mirror similar to Mission Planner?

ShiftingKill, 2024-10-03:
Mission Planner has a straightforward way to mirror packets received by the GCS on the local network, ie the PC running Mission Planner can also have a script running that sees the MAVLINK messages on the network. You simply connect to a flight vehicle, the packets start coming in, and my script can pick up on the localhost-mirrored packets. I am creating a GPS-based gimbal tracking script that tracks a flight vehicle and the script processes the MAVLINK messages to determine lat, lon, and alt. I successfully implemented this with Mission Planner but when converting to QGC I cannot seem to pick up the packets on the local network. I'm not sure if there is something I'm missing but I followed the instructions for MAVLINK message forwarding, but I think that assumes another instance of QGC is listening, it isn't really the same as Mission Planner's mirroring. How can we get these raw packets from QGC to the local network?

I do know that the solution should not involve a different library or MAVPROXY, this is something that QGC should do by default just like Mission Planner does it, we just need access to the packets. I will PR if necessary to implement this feature.

Paschalis, 2024-10-03:
It sounds like you're looking for a way to get access to the raw MAVLink packets in QGroundControl (QGC), similar to how Mission Planner mirrors them on the local network for use in external scripts like your gimbal tracking script.

### Here's the good news:
QGroundControl already has built-in MAVLink message forwarding.

> **Note**: "Enable MAVLink forwarding: Enable QGroundControl to forward MAVLink messages from connected vehicles to another UDP endpoint (Default: False). Note that this forwarding is one-way: from QGC to the specified host. Any MAVLink messages that are received from the specified host will be ignored." (Source: [QGroundControl Documentation](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/settings_view/mavlink.html#ground_station))

This forwarding is quite similar to what you're used to in Mission Planner. It doesn't just assume another instance of QGC is listening—it can forward packets to any application or script that listens on the specified port. 

## Here's what you need to do:

1. **Enable MAVLink Message Forwarding**:
   - In QGC, go to **Settings > General > MAVLink**.
   - Enable **MAVLink message forwarding**. 
   - Set the **Forwarding address** to `localhost:14550` (or another port of your choice). 

2. **Listen on the Forwarded Port**:
   Your gimbal tracking script just needs to listen on the port where QGC is forwarding the MAVLink packets (e.g., `localhost:14550`). 

   Here’s a basic example in Python:
   ```python
   import socket

   # UDP socket to listen for MAVLink packets
   udp_socket = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
   udp_socket.bind(('localhost', 14550))  # Port matches QGC forwarding setting

   while True:
       data, addr = udp_socket.recvfrom(1024)  # Buffer size 1024 bytes
       # Process MAVLink data
       print(data)

Paschalis, 2024-10-03:
Since QGC already has the **MAVLink message forwarding** feature, there’s no need to introduce additional routing tools like **MAVProxy** or **MAVLink Router**. QGC forwards the packets directly, and your script can pick them up on the specified port.

closed 2024-10-03 by ShiftingKill