mavlink/qgroundcontrol #11966: Follow failed: Home position not set

twistedwings, 2024-10-03:
If "Stream GCS Position" is enabled in Application settings, the message "Follow failed: Home position not set" pops up.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: 
[image]
- QGC build: daily
- Flight Controller: SITL
- Autopilot (with version): Ardupilot Copter 4.6.0

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
[image]

HTRamsey, 2024-10-06:
This will probably happen until takeoff when you get your new home position

twistedwings, 2024-10-06:
The code looks like it's related to follow mode.
In the stable version is an option Stream GCS Position-only in follow mode.
In the daily the option is on/off only.
So there shouldn't be any dependencies to follow.

twistedwings, 2024-10-07:
I remember an years lasting [discusion](https://github.com/mavlink/qgroundcontrol/issues/7811) about follow mode with the Ardupilot folks.
I think the code uses the home altitude of the vehicle instead of gps altitude.
Assuming gcs and vehicle are both at the same altitude, which is true in most cases.
The reason for this trick is that the altitude information is garbage on most smartphones.
So the code makes sense, only the error message is confusing and should be changed in "waiting for home" or the like.

HTRamsey, 2024-10-07:
first thing to do is to restore the settings #11975

DonLakeFlyer, 2024-10-09:
> I think the code uses the home altitude of the vehicle instead of gps altitude.
> Assuming gcs and vehicle are both at the same altitude, which is true in most cases.
> The reason for this trick is that the altitude information is garbage on most smartphones.

Correct.

The original pull is here: https://github.com/mavlink/qgroundcontrol/pull/8350.

DonLakeFlyer, 2024-10-09:
> So the code makes sense, only the error message is confusing and should be changed in "waiting for home" or the like.

I'll take a look...