mavlink/qgroundcontrol #11968: Undefined symbols for architecture arm64:

Prabha10cs, 2024-10-04:
Undefined symbols for architecture arm64:
  "qt_static_plugin_QGeoServiceProviderFactoryQGC()", referenced from:
      main::StaticQGeoServiceProviderFactoryQGCPluginInstance::StaticQGeoServiceProviderFactoryQGCPluginInstance() in main.cc.o
  "QGCApplication::init()", referenced from:
      _main in main.cc.o
  "QGCApplication::shutdown()", referenced from:
      _main in main.cc.o
  "QGCApplication::QGCApplication(int&, char**, bool)", referenced from:
      _main in main.cc.o
  "QGCApplication::~QGCApplication()", referenced from:
      _main in main.cc.o
      _main in main.cc.o
  "QGC::initTimer()", referenced from:
      _main in main.cc.o
  "RunGuard::tryToRun()", referenced from:
      _main in main.cc.o
  "RunGuard::RunGuard(QString const&)", referenced from:
      _main in main.cc.o
ld: symbol(s) not found for architecture arm64
c++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [QGroundControl.app/Contents/MacOS/QGroundControl] Error 1
make[1]: *** [CMakeFiles/QGroundControl.dir/all] Error 2
make: *** [all] Error 2

HTRamsey, 2024-10-04:
What setup gave you this?

HTRamsey, 2024-10-10:
I am going to link this to your latest issue #11992

closed 2024-10-10 by HTRamsey