mavlink/qgroundcontrol #11978: "Upload Required" after mission download without changes

twistedwings, 2024-10-07:
After a mission download the message "Upload Required" appears although nothing was changed.

## Steps to Reproduce:
1. download a mission

## System Information
- Operating System: [Windows 10]
- QGC Version: [4.3.0-876]
- QGC build: [daily]
- Flight Controller: [SITL]
- Autopilot (with version): [Ardupilot Copter 4.6.0]

## Log Files and Screenshots
[image]

DonLakeFlyer, 2024-10-13:
Can you provide some more details steps to repro. When I try it on daily it's working fine.

DonLakeFlyer, 2024-10-13:
Hmm, might be ok on OSX as well...

twistedwings, 2024-10-14:
It's still there in 4.3.0-971.
If there's a mission on the vehicle, QGC downloads the mission on connect and says "Upload Required" in plan view.