mavlink/qgroundcontrol #11984: Build gives error when CMake file is modified

serkanMzlm, 2024-10-08:
Environment and versions I work with:

Windows 10
QT: 6.6.3
Problem: When I try to build the QGC application via cmake, it compiles the first time and runs the application. However, when I make a change on cmake, it gives an error. When I delete the build file and compile it again, it works this time.

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
C:/ems/qgroundcontrol/build/Desktop_Qt_6_6_3_MSVC2019_64bit-Release/_deps/exiv2-src/src/ZLIB_INCLUDE_DIR
   used as include directory in directory C:/ems/qgroundcontrol/build/Desktop_Qt_6_6_3_MSVC2019_64bit-Release/_deps/exiv2-src/src

CMake Warning (dev) in build/Desktop_Qt_6_6_3_MSVC2019_64bit-Release/_deps/geographiclib-src/src/CMakeLists.txt:
  Policy CMP0100 is not set: Let AUTOMOC and AUTOUIC process .hh files.  Run
  "cmake --help-policy CMP0100" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  For compatibility, CMake is excluding the header file(s):

    "C:/ems/qgroundcontrol/build/Desktop_Qt_6_6_3_MSVC2019_64bit-Release/_deps/geographiclib-src/src/kissfft.hh"

  from processing by AUTOMOC and AUTOUIC.  If any of the files should be
  processed, set CMP0100 to NEW.  If any of the files should not be
  processed, explicitly exclude them by setting the source file property
  SKIP_AUTOGEN:

    set_property(SOURCE file.hh PROPERTY SKIP_AUTOGEN ON)

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done (1.9s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
ninja: error: rebuilding 'build.ninja': subcommand failed
FAILED: build.ninja 
C:\Qt\Tools\CMake_64\bin\cmake.exe --regenerate-during-build -SC:\ems\qgroundcontrol -BC:\ems\qgroundcontrol\build\Desktop_Qt_6_6_3_MSVC2019_64bit-Release
11:21:30: The process "C:\Qt\Tools\CMake_64\bin\cmake.exe" exited with code 1.
Error while building/deploying project QGroundControl (kit: Desktop Qt 6.6.3 MSVC2019 64bit)
When executing step "Build"
```

I just added `message(“aaaa”)" to the cmake file

HTRamsey, 2024-10-08:
I don't know but in this particular case it looks like it can't find zlib which is weird because the part of exiv2 that uses zlib (png compatibility) is currently disabled. I just looked at their cmake and it does seem they have one spot where they try to link to zlib regardless of if png compatibility is enabled. Maybe you can mess around with declaring ZLIB_INCLUDE_DIR or trying a different version of exiv2

serkanMzlm, 2024-10-09:
@HTRamsey  The problem is that when I delete the build file and build again it works. But cmake gives this error as soon as I add anything.

HTRamsey, 2024-10-09:
I got a PR committed in exiv2 to fix that error

closed 2024-11-30 by HTRamsey