mavlink/qgroundcontrol #11988: build process has failed

Prabha10cs, 2024-10-09:
:-1: error: linker command failed with exit code 1 (use -v to see invocation)
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

[image]

[image]

Prabha10cs, 2024-10-09:
after building, I tried to run the emulator that time, the above error showed

HTRamsey, 2024-10-09:
You didn't show the actual error that causes the build to terminate though, this tells me nothing so I can't help

Prabha10cs, 2024-10-09:
[image]
this is my actual error 
after building the project, I tried to run the emulator that time, and the above error showed

HTRamsey, 2024-10-09:
probably there is more info elsewhere in the build/configure log

Prabha10cs, 2024-10-10:
FAILED: _deps/libevents-build/parse/liblibevents_health_and_arming_checks.so 
: && /Users/prabhakaran/Library/Android/sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android23 --sysroot=/Users/prabhakaran/Library/Android/sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -fPIC -DQT_QML_DEBUG   -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fexceptions -frtti -stdlib=libc++ -g  -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--gc-sections -Qunused-arguments -Wl,--no-undefined -shared -Wl,-soname,liblibevents_health_and_arming_checks.so -o _deps/libevents-build/parse/liblibevents_health_and_arming_checks.so _deps/libevents-build/parse/CMakeFiles/libevents_health_and_arming_checks.dir/libevents_health_and_arming_checks_autogen/mocs_compilation.cpp.o _deps/libevents-build/parse/CMakeFiles/libevents_health_and_arming_checks.dir/health_and_arming_checks.cpp.o  -latomic -lm && :
ld: error: undefined symbol: events::parser::ParsedEvent::argumentValueInt(int) const
>>> referenced by health_and_arming_checks.cpp:47 (_deps/libevents-src/libs/cpp/parse/health_and_arming_checks.cpp:47)
>>>               _deps/libevents-build/parse/CMakeFiles/libevents_health_and_arming_checks.dir/health_and_arming_checks.cpp.o:(events::HealthAndArmingChecks::handleEvent(events::parser::ParsedEvent const&))
>>> referenced by health_and_arming_checks.cpp:50 (_deps/libevents-src/libs/cpp/parse/health_and_arming_checks.cpp:50)
>>>               _deps/libevents-build/parse/CMakeFiles/libevents_health_and_arming_checks.dir/health_and_arming_checks.cpp.o:(events::HealthAndArmingChecks::handleEvent(events::parser::ParsedEvent const&))
>>> referenced by health_and_arming_checks.cpp:51 (_deps/libevents-src/libs/cpp/parse/health_and_arming_checks.cpp:51)
>>>               _deps/libevents-build/parse/CMakeFiles/libevents_health_and_arming_checks.dir/health_and_arming_checks.cpp.o:(events::HealthAndArmingChecks::handleEvent(events::parser::ParsedEvent const&))
>>> referenced 8 more times

ld: error: undefined symbol: events::parser::ParsedEvent::message() const
>>> referenced by health_and_arming_checks.cpp:85 (_deps/libevents-src/libs/cpp/parse/health_and_arming_checks.cpp:85)
>>>               _deps/libevents-build/parse/CMakeFiles/libevents_health_and_arming_checks.dir/health_and_arming_checks.cpp.o:(events::HealthAndArmingChecks::handleEvent(events::parser::ParsedEvent const&))

ld: error: undefined symbol: events::parser::ParsedEvent::description() const
>>> referenced by health_and_arming_checks.cpp:86 (_deps/libevents-src/libs/cpp/parse/health_and_arming_checks.cpp:86)
>>>               _deps/libevents-build/parse/CMakeFiles/libevents_health_and_arming_checks.dir/health_and_arming_checks.cpp.o:(events::HealthAndArmingChecks::handleEvent(events::parser::ParsedEvent const&))
clang++: error: linker command failed with exit code 1 (use -v to see invocation)

how to solve this problem @HTRamsey

HTRamsey, 2024-10-10:
I am going to link this to your latest issue #11992