mavlink/qgroundcontrol #11996: [Bug] QGC not showing external mode from PX4

junwoo091400, 2024-10-11:
## Expected Behavior
QGC should show the name of the 'external mode' supported by PX4 via it's [ROS2 Control interface](https://docs.px4.io/main/en/ros2/px4_ros2_control_interface.html). This feature was added in https://github.com/mavlink/qgroundcontrol/pull/10788.

## Current Behavior
The mode name doesn't show up in flight modes supported.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Install [QGC v4.4.2](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.4.2)
2. Start PX4 SITL `make px4_sitl_default gz_x500`
3. Start MicroXRCE-DDS Agent `MicroXRCEAgent udp4 -p 8888`
4. Start ROS2 script `ros2 run custom_mode custom_mode`
5. Check QGC's listed supported modes

## System Information
* OS: Ubuntu 22.04.3 LTS
* PX4: [commit 8070c70](https://github.com/PX4/PX4-Autopilot/commit/8070c70)
* ROS2 program from https://github.com/ARK-Electronics/px4_ros2_examples_ws?tab=readme-ov-file

## Log Files and Screenshots
[image]

[image]

With the "StandardModes" for QGC Console log enabled, it didn't show anything. Only irrelevant errors:
[image]

patrikpordi, 2024-10-29:
I face the same issue!

DonLakeFlyer, 2024-10-29:
Can you try a daily build as well?

patrikpordi, 2024-10-29:
It works on the daily build but based on the https://github.com/mavlink/qgroundcontrol/pull/10788 I assume it should work on the main as well? Do we misunderstand it?

hamishwillee, 2024-10-29:
This _should_ work by requesting [AVAILABLE_MODES](https://mavlink.io/en/messages/development.html#AVAILABLE_MODES)  and then populating the UI appropriately. That message is in development.xml - does QGC import this for both main and daily builds?

DonLakeFlyer, 2024-11-15:
@bkueng I assume this is all through the new mavlink flight mode stuff Beat added? Sorry I don't know much about that. Should that be working in Stable?

patrikpordi, 2024-11-18:
It only works in daily build at the moment. Are you planning on adding it to the main anytime soon?

bkueng, 2024-11-19:
> Should that be working in Stable?

It's disabled for releases as the messages are in `development.xml` and thus should not end up in releases. I think making it a runtime option (disabled by default for releases) would also be a possibility.
I don't know if or how that changed with the switch to the cmake build.

DonLakeFlyer, 2024-11-19:
Ah, ok Thanks for the info Beat. So this is working as expected.

closed 2024-11-19 by DonLakeFlyer