mavlink/qgroundcontrol #11997: CMake failure when using VS 2022 as generator

NEIL-smtg, 2024-10-11:
Recently, as I tried to build QGC from source, I encountered the following error in cmake:

```
-- Viewer3D is Initialized
CMake Error at C:/gitP/Microsoft/vcpkg/installed/x64-windows/share/Qt6/QtPublicWalkLibsHelpers.cmake:259 (message):
  The Quick3D target is mentioned as a dependency for Viewer3D, but not
  declared.
Call Stack (most recent call first):
  C:/gitP/Microsoft/vcpkg/installed/x64-windows/share/Qt6/QtPublicWalkLibsHelpers.cmake:228 (__qt_internal_walk_libs)
  C:/gitP/Microsoft/vcpkg/installed/x64-windows/share/Qt6/QtPublicWalkLibsHelpers.cmake:311 (__qt_internal_walk_libs)
  C:/gitP/Microsoft/vcpkg/installed/x64-windows/share/Qt6Core/Qt6CoreMacros.cmake:651 (__qt_internal_collect_all_target_dependencies)
  C:/gitP/Microsoft/vcpkg/installed/x64-windows/share/Qt6Core/Qt6CoreMacros.cmake:774 (_qt_internal_finalize_executable)
  C:/gitP/Microsoft/vcpkg/installed/x64-windows/share/Qt6Core/Qt6CoreMacros.cmake:590:EVAL:1 (qt6_finalize_target)
  CMakeLists.txt:DEFERRED
```

## Expected Behavior
No issue.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. open x64 native tools command prompt windows for vs2022
2. git clone https://github.com/microsoft/vcpkg.git
3. cd microsoft\vcpkg
4. bootstrap-vcpkg.bat 2>&1
5. vcpkg.exe install --x-buildtrees-root=C:\vcpkgbt --keep-going --recurse qtcharts qtlocation qtpositioning qtspeech qt5compat qtmultimedia qtserialport qtimageformats qtshadertools qtconnectivity qtquick3d qtsensors --triplet x64-windows --clean-after-build 2>&1
6. C:\tools\msys64\usr\bin\wget https://gstreamer.freedesktop.org/data/pkg/windows/1.24.8/msvc/gstreamer-1.0-msvc-x86_64-1.24.8.msi --tries=5 --retry-connrefused --retry-on-host-error --waitretry=10 -O C:\gstreamer\gstreamer-1.0-msvc-x86_64-1.24.8.msi 2>&1
7. C:\tools\msys64\usr\bin\wget https://gstreamer.freedesktop.org/data/pkg/windows/1.24.8/msvc/gstreamer-1.0-devel-msvc-x86_64-1.24.8.msi --tries=5 --retry-connrefused --retry-on-host-error --waitretry=10 -O C:\gstreamer\gstreamer-1.0-devel-msvc-x86_64-1.24.8.msi 2>&1
8. cd /d C:\gstreamer
9. msiexec /passive INSTALLDIR=C:\gstreamer ADDLOCAL=ALL /i 
10. msiexec.exe /passive INSTALLDIR=C:\gstreamer ADDLOCAL=ALL /i 
11. set PATH=C:\gstreamer\1.0\msvc_x86_64\bin;%PATH%
12. set PKG_CONFIG_PATH=C:\gstreamer\1.0\msvc_x86_64\lib;C:\gstreamer\1.0\msvc_x86_64\lib\pkgconfig;C:\gstreamer\1.0\msvc_x86_64\lib\gstreamer-1.0\pkgconfig;%PKG_CONFIG_PATH%
13. cd c:\
14. git clone https://github.com/mavlink/qgroundcontrol.git
15. mkdir mavlink\qgroundcontrol\build_amd64
16. cd mkdir mavlink\qgroundcontrol\build_amd64
17. cmake -G "Visual Studio 17 2022" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.22621.0 -DCMAKE_TOOLCHAIN_FILE=C:\gitP\Microsoft\vcpkg\scripts\buildsystems\vcpkg.cmake -DVCPKG_TARGET_TRIPLET=x64-windows -DCMAKE_BUILD_TYPE=release -DQGC_STABLE_BUILD=OFF -DQGC_ENABLE_QT_VIDEOSTREAMING=OFF .. 2>&1

Build log: [qg.log](https://github.com/user-attachments/files/17341082/qg.log)

## System Information
When posting bug reports, include the following information
- Operating System: Windows 11
- QGC Version: latest commit
- QGC build: self-built from source
- Flight Controller: -
- Autopilot (with version): -
- QT Version: 6.7.2

HTRamsey, 2024-10-11:
So QGC hasn't been tested on 6.7.2, and 6.6.3 doesn't support 2022. But regardless the issue you're having is unrelated to that and was fixed in #11960 I think

NEIL-smtg, 2024-10-14:
@HTRamsey Thanks for your response. I realized that I made an error with the commit SHA. I have tested locally, and the build succeeded. My apologies for the mistake and I'll go ahead and close this issue.

closed 2024-10-14 by NEIL-smtg