mavlink/qgroundcontrol #11999: OSX daily builds crash on startup

DonLakeFlyer, 2024-10-12:
Install latest OSX daily build from web site. Boot and it will crash on startup:
```
Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @rpath/libexiv2.28.dylib
Referenced from: <4E27540A-BD22-3B17-8130-ECD2F52D9E22> /Applications/QGroundControl Daily.app/Contents/MacOS/QGroundControl
Reason: tried: '/Library/Frameworks/GStreamer.framework/Versions/Current/lib/pkgconfig/../../lib/libexiv2.28.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/GStreamer.framework/Versions/Current/lib/pkgconfig/../../lib/libexiv2.28.dylib' (no such file), '/Applications/QGroundControl Daily.app/Contents/Frameworks/libexiv2.28.dylib' (no such file), '/Library/Frameworks/GStreamer.framework/Versions/Current/lib/pkgconfig/../../lib/libexiv2.28.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/GStreamer.framework/Versions/Current/lib/pkgconfig/../../lib/libexiv2.28.dylib' (no such file), '/Applications/QGroundControl Daily.app/Contents/Frameworks/libexiv2.28.dylib' (no such file)
(terminated at launch; ignore backtrace)
```

DonLakeFlyer, 2024-10-12:
Tried building master without gstreamer and that doesn't seem to work any more:
```
CMake Error at /usr/local/Cellar/cmake/3.29.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
[cmake]   Could NOT find GStreamer (missing: GStreamer_Core_FOUND
[cmake]   GStreamer_Base_FOUND GStreamer_Video_FOUND GStreamer_Gl_FOUND Allocators
[cmake]
```

Used to be that build would detect no gstreamer and build a version automatically without it.

DonLakeFlyer, 2024-10-12:
I tried going to the dev guide for building to see what the new rules are but that doesn't seem to be current. This "Video Streaming/Gstreamer: - see [Video Streaming](https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md)." just goes to a 404 now.

Not sure what the current state of affairs is with respect to gstreamer installs.

HTRamsey, 2024-10-12:
@DonLakeFlyer these errors all mention exiv2. Maybe I should revert to before using that

HTRamsey, 2024-10-12:
Actually nevermind it gets built as a shared lib then doesn't get copied over when deployed. Also in regards to gstreamer, yeah I guess currently you would have to disable build with gstreamer in the options.

DonLakeFlyer, 2024-10-13:
Ah, I saw it poking around in the gstreamer framework! That it was that.

> but I also feel like they should have to disable building with gstreamer to acknowledge it won't have it, rather than just be surprised because they didn't read the configure output log.

Make sense. Can you update the dev guide with respect to how you are supposed to build with/without gstreamer?

HTRamsey, 2024-10-14:
Is this resolved now with the static lib thing?

DonLakeFlyer, 2024-10-14:
Tried daily again:
```
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @rpath/libavcodec.59.18.100.dylib
Referenced from: <6B366159-8F7A-39AE-A6C9-B2404185194D> /Applications/QGroundControl Daily.app/Contents/MacOS/QGroundControl
Reason: tried: '/Library/Frameworks/GStreamer.framework/Versions/Current/lib/pkgconfig/../../lib/libavcodec.59.18.100.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/GStreamer.framework/Versions/Current/lib/pkgconfig/../../lib/libavcodec.59.18.100.dylib' (no such file), '/Applications/QGroundControl Daily.app/Contents/Frameworks/libavcodec.59.18.100.dylib' (no such file), '/Library/Frameworks/GStreamer.framework/Versions/Current/lib/pkgconfig/../../lib/libavcodec.59.18.100.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/GStreamer.framework/Versions/Current/lib/pkgconfig/../../lib/libavcodec.59.18.100.dylib' (no such file), '/Applications/QGroundControl Daily.app/Contents/Frameworks/libavcodec.59.18.100.dylib' (no such file)
(terminated at launch; ignore backtrace)
```

HTRamsey, 2024-10-14:
Oof, okay. apparently something else not being copied over. I'll check it out tonight

HTRamsey, 2024-10-14:
Can you try an artifact from #11971? I think that should fix this

DonLakeFlyer, 2024-10-14:
Whack-a-mole :)

```
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @rpath/libgstcoreelements.dylib
Referenced from: <F33C7B3E-18F8-3E0A-AC9F-70A6E41A4A77> /Applications/QGroundControl Daily.app/Contents/MacOS/QGroundControl
Reason: tried: '/Library/Frameworks/GStreamer.framework/Versions/1.0/lib/pkgconfig/../../lib/libgstcoreelements.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/GStreamer.framework/Versions/1.0/lib/pkgconfig/../../lib/libgstcoreelements.dylib' (no such file), '/Applications/QGroundControl Daily.app/Contents/Frameworks/libgstcoreelements.dylib' (no such file), '/Library/Frameworks/GStreamer.framework/Versions/1.0/lib/pkgconfig/../../lib/libgstcoreelements.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/GStreamer.framework/Versions/1.0/lib/pkgconfig/../../lib/libgstcoreelements.dylib' (no such file), '/Applications/QGroundControl Daily.app/Contents/Frameworks/libgstcoreelements.dylib' (no such file)
(terminated at launch; ignore backtrace)
```

HTRamsey, 2024-10-15:
So the frustrating part is setting the gstreamer environment variables based on whether you are just running a build or are running from an installation, because for the first case you want to pull from the system's gstreamer and in the second case you want to pull from the app's included gstreamer. Basically if you have it set up to work one way, it won't work the other way. Note the ifdef QGC_INSTALL_RELEASE [here](https://github.com/mavlink/qgroundcontrol/blob/be1e2d46c95812d3ad700bdc4807eca95829be6b/src/VideoManager/VideoReceiver/GStreamer/GStreamer.cc#L173). For an appimage on linux, you can define these environment variables in a custom script that gets run at the start of the appimage. However for MacOS and DMG's, I have no idea how to approach this

HTRamsey, 2024-10-16:
Yeah, just trying to keep it as automated and simple as possible. I'm hoping I solved this in master now, apparently you can set environment variables in the plist file so that's a potential solution as well. I really just need to get a MacOS system from ebay or something so it's easier for me to test

closed 2025-03-17 by HTRamsey