mavlink/qgroundcontrol #12002: ADSB_Simulator.py Script Not Generating Vehicles on the Map

Paschalis, 2024-10-12:
## Expected Behavior  
Should generate ADS-B messages for aircraft, allowing them to appear on the map in QGroundControl.

[image]

## Current Behavior  
When running the default version of `qgroundcontrol/test/ADSB/ADSB_Simulator.py`, no vehicles appear on the map despite the server running.

## Steps to Reproduce:  
1. Clone the QGroundControl repository.
2. Navigate to `qgroundcontrol/test/ADSB/`.
3. Run the ADSB simulator with the command:  
    ```bash
    python3 ADSB_Simulator.py
    ```
5. Open QGroundControl and connect to the ADS-B feed (default host: localhost and port: 30003).
6. Observe that no vehicles are displayed on the map.

## System Information
- Operating System: Ubuntu 24.04.1 LTS
- QGC Version: Daily Version

HTRamsey, 2024-10-12:
I was just using that script for some unit tests when working on ADSB, it doesn't actually send enough data to currently display anything in QGC

Paschalis, 2024-10-12:
Hi @HTRamsey,

Thank you for your feedback! I understand that the original script didn't send enough data for effective testing in QGroundControl. In this update, I've enhanced the script to generate and send sufficient ADS-B messages, which allows vehicles to be displayed correctly on the map.

This improvement is especially useful for users who may not have access to SDR dongle hardware, as it provides a way to simulate ADS-B data without needing **physical equipment**. It should facilitate easier testing and development in QGC.

HTRamsey, 2024-10-12:
Nice, thanks for doing that

DonLakeFlyer, 2024-10-13:
MockLinks already support ADSB vehicles for testing without ADSB hardware.

Paschalis, 2024-10-13:
Thank you for the feedback, @DonLakeFlyer!

After your suggestion, I revisited the [Mock Link documentation](https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/tools/mock_link.html#mock-link) in the QGroundControl Developer Guide. While testing, I observed that adding a PX4 or Generic Vehicle through Mock Link only creates a single ADS-B plane on the map. However, the behavior of this plane appears unusual—it moves erratically, *jumping back and forth between random points without smooth transitions*. Additionally, the heading seems "frozen" and doesn’t consistently align with the flight course, making it difficult to simulate realistic scenarios.

This is why I believe the updated `ADSB_Simulator.py` script in [PR #12003](https://github.com/mavlink/qgroundcontrol/pull/12003) can be valuable. It can provide more consistent and realistic flight paths, making it useful for debugging and **feature testing**.

DonLakeFlyer, 2024-10-13:
So fix the ADSB support in MockLink to be better  instead of creating something which duplicates already existing functionality.

DonLakeFlyer, 2024-10-13:
@HTRamsey @Paschalis Is there a reason why an external script would be better than the internal MockLink support? Don't want to have to manage things in two places. We should pick one. In general MockLink is the place to add useful testing oriented things so you don't need any hardware/sitl.

Paschalis, 2024-10-13:
@DonLakeFlyer  I understand your concern about avoiding duplicate functionality. I wasn't aware that MockLink already supported ADSB vehicles when I initially worked on the `ADSB_Simulator.py` script. My main goal was to investigate and address the issue of ground speed showing as NaN knots on ADSB vehicles, and using the script helped me simulate ADSB data without needing to rely on my SDR dongle and dump1090 setup.

I would definitely prefer to improve the existing MockLink ADSB support, especially now that I understand its potential for testing without requiring hardware. It might take me a bit of time to implement those improvements. In the meantime, if it makes sense, merging this script update could serve as a useful temporary solution for anyone looking to simulate more consistent and realistic ADSB behavior.

HTRamsey, 2024-10-13:
Personally I have no preference, I just didn't want to risk messing up mocklink in any way when I just wanted a quick adsb test. That script actually just came from chatgpt anyways.

DonLakeFlyer, 2024-10-13:
Ok, let's move the better adsb script support into MockLink then. as the long term solution. I'll merge the script updates for now.

PR #12003 (2024-10-13): Fix ADSB_Simulator.py to generate vehicles on the map (#12002)
Description
-----------
This pull request addresses the issue where vehicles were not appearing on the map in QGroundControl when using the ADSB_Simulator.py script. The script has been updated to correctly generate and send ADS-B messages.

[image]

Checklist:
----------

- [x] [Review Contribution Guidelines](https://github.com/mavlink/qgroundcontrol/blob/master/.github/CONTRIBUTING.md).
- [x] [Review Code of Conduct](https://github.com/mavlink/qgroundcontrol/blob/master/.github/CODE_OF_CONDUCT.md).
- [x] I have tested my changes [... cut]
closed 2024-10-13 by DonLakeFlyer