mavlink/qgroundcontrol #12014: Send stream from camera to RTMP/RTSP server

antonymarion, 2024-10-15:
Hello,

Does the QGC API provide a way to redistribute stream from UAV's camera to a RTMP/RTSP server?

Regards,

closed 2024-12-05 by antonymarion

tranhien1612, 2025-05-08:
@antonymarion Did you do this problem?

antonymarion, 2025-05-10:
@tranhien1612 Yes.

If you need any assistance/support on this feature you can book a meeting on the bottom page of https://corner-stone.ai/ "Réserver un créneau dans notre agenda"

Regards
Antony

tranhien1612, 2025-05-16:
@antonymarion can you show you this feature?

antonymarion, 2025-05-16:
Yes.

Please book a meeting in my agenda

tranhien1612, 2025-05-16:
@antonymarion i think i don't enjoy your metting. Did you try it in android system and what version of QT are you using?. If possible can I have your source code?

antonymarion, 2025-05-16:
I can provide you support as a paid service; Unfortunately I won't provide source code for this feature.