mavlink/qgroundcontrol #12022: GStreamer doesn't work in the daily builds

eshcheglov, 2024-10-17:
## Steps to Reproduce:
1. Run latest QGC Daily build (latest-2-gfe4f8a42c 64 bit)
2. Add RTSP video stream: Application Settings -> Video -> Connection -> RTSP URL

## System Information
- Operating System: SteamOS Holo (Arch Linux)
- QGC Version: latest-2-gfe4f8a42c 64 bit
- QGC build: daily

Stable version of QGC works well

## Log Files and Screenshots
[QGC.log](https://github.com/user-attachments/files/17415009/QGC.log)

HTRamsey, 2024-10-17:
Should be good after the linked PR is merged. I had to mark the plugin scanner as executable

eshcheglov, 2024-10-18:
Thanks!

I tried to repack the existing AppImage and made the following changes:

- Used the AppRun script from your branch.
- Added chmod +x to gst-plugin-scanner and gst-ptp-helper.

Now I see the following in the `VideoReceiverLog: "GStreamer error: Could not initialize window system.` 
I suppose the changes in the source code that you made in your branch (like in gstreamer.cc)  were aimed specifically at fixing this error?

HTRamsey, 2024-10-18:
Yeah it's honestly been a pain. In older versions QGC just didn't provide any dependencies in the AppImage which I suppose I could just give up and do too but I'd really like to get this working well

HTRamsey, 2024-10-18:
@eshcheglov is your build finding gstreamer-gl-x11-1.0?

eshcheglov, 2024-10-18:
There is no errors about `gstreamer-gl-x11-1.0` in my log file:

```bash
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
Error loading text-to-speech plug-in "speechd"
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
No engine set.
qgc.audio.audiooutput: void AudioOutput::say(const QString&, AudioOutput::TextMods) Speech Not Supported: " position flight mode"
ComponentInformationTranslationLog: Locale  "en_US"  not found in json
ComponentInformationTranslationLog: Locale  "en_US"  not found in json
ParameterManagerLog: Attemping load from cache
ParameterManagerLog: Parameters cache match failed /home/deck/.config/QGroundControl.org/ParamCache/1_1.v2
VideoReceiverLog: GStreamer error: Could not initialize window system
VideoReceiverLog: GStreamer error: Could not initialize window system
VideoReceiverLog: GStreamer error: Could not initialize window system
VideoReceiverLog: GStreamer error: Could not initialize window system
```

`libgstgl` (I suppose that `gstreamer-gl-x11-1.0` is `libgstgl`) is present both in my system and in the AppImage:
- AppImage: usr/lib/libgstgl-1.0.so.0
- My rootfs: usr/lib/libgstgl-1.0.so.0.2203.0 (from gst-plugins-base-libs 1.22.3-6)

HTRamsey, 2024-10-19:
I did a quick temporary fix in #12029, there's still some issue with audio but I'll get a better solution out soon.

closed 2024-10-19 by HTRamsey