mavlink/qgroundcontrol #12024: Could NOT find GStreamer (missing: Fft)

david6096, 2024-10-18:
When I build the qgroundcontorl master version for android arm64,cmake errors:

CMake Error at /home/star/Qt/Tools/CMake/share/cmake-3.29/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
[cmake]   Could NOT find GStreamer (missing: Fft)

some tips for this?

thanks

HTRamsey, 2024-10-18:
I think there was a typo in FindGStreamer.cmake. Can you find this code and fix the line that starts with HEADER and try that?
[image]

david6096, 2024-10-18:
yes,has gone!!
thanks a lot

david6096, 2024-10-18:
another question:
build error:
FAILED: libQGroundControl_arm64-v8a.so 
: && ccache /home/star/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android23 --sysroot=/home/star/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fexceptions -frtti -stdlib=libc++ -O3 -DNDEBUG  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--gc-sections -Qunused-arguments -Wl,--no-undefined   -Wl,-Bsymbolic -shared  -o libQGroundControl_arm64-v8a.so CMakeFiles/QGroundControl.dir/QGroundControl_autogen/mocs_compilation.cpp.o CMakeFiles/QGroundControl.dir/src/main.cc.o CMakeFiles/QGroundControl.dir/build/Qt_6_6_3_Clang_arm64_v8a-release/.rcc/qrc_QGroundControl_translations.cpp.o CMakeFiles/QGroundControl.dir/QGroundControl_autogen/EWIEGA46WW/qrc_qgcimages.cpp.o CMakeFiles/QGroundControl.dir/QGroundControl_autogen/EWIEGA46WW/qrc_qgcresources.cpp.o CMakeFiles/QGroundControl.dir/QGroundControl_autogen/EWIEGA46WW/qrc_qgroundcontrol.cpp.o CMakeFiles/QGroundControl.dir/QGroundControl_autogen/RBUUZA67W2/qrc_InstrumentValueIcons.cpp.o CMakeFiles/QGroundControl.dir/QGroundControl_autogen/ZHQSJ6LXPM/qrc_APMResources.cpp.o CMakeFiles/QGroundControl.dir/QGroundControl_autogen/22EX5NVWOJ/qrc_PX4Resources.cpp.o CMakeFiles/QGroundControl.dir/QGroundControl_autogen/OMJOBHPZGB/qrc_utmsp_dummy.cpp.o CMakeFiles/QGroundControl.dir/android/src/AndroidInit.cpp.o -L/home/star/qgroundcontrol/build/Qt_6_6_3_Clang_arm64_v8a-release/_deps/gstreamer/install/gstreamer/arm64/lib   -L/home/star/qgroundcontrol/build/Qt_6_6_3_Clang_arm64_v8a-release/_deps/gstreamer/install/gstreamer/arm64/lib/gstreamer-1.0 src/QtLocationPlugin/CMakeFiles/QGCLocation_init.dir/./QGCLocation_init_autogen/mocs_compilation.cpp.o  src/QtLocationPlugin/CMakeFiles/QGCLocation_init.dir/./QGCLocation_init.cpp.o  src/AutoPilotPlugins/PX4/CMakeFiles/PX4AutoPilotPlugin_resources_1.dir/./.rcc/qrc_PX4AutoPilotPlugin_res.cpp.o  /home/star/Qt/6.6.3/android_arm64_v8a/lib/libQt6Svg_arm64-v8a.so  src/libQGC_arm64-v8a.a  src/QmlControls/libQmlControls_arm64-v8a.a  src/Utilities/libUtilities_arm64-v8a.a  src/API/libAPI_arm64-v8a.a  src/ADSB/libADSB_arm64-v8a.a  src/AnalyzeView/libAnalyzeView_arm64-v8a.a  src/Audio/libAudio_arm64-v8a.a  src/AutoPilotPlugins/libAutoPilotPlugins_arm64-v8a.a  src/Camera/libCamera_arm64-v8a.a  src/Comms/libComms_arm64-v8a.a  src/AutoPilotPlugins/Common/libCommonAutoPilotPlugin_arm64-v8a.a  src/FactSystem/libFactSystem_arm64-v8a.a  src/FirmwarePlugin/libFirmwarePlugin_arm64-v8a.a  src/FollowMe/libFollowMe_arm64-v8a.a  src/Gimbal/libGimbal_arm64-v8a.a  src/GPS/libGPS_arm64-v8a.a  src/VideoManager/VideoReceiver/GStreamer/libGStreamerReceiver_arm64-v8a.a  src/Joystick/libJoystick_arm64-v8a.a  src/MAVLink/libMAVLink_arm64-v8a.a  src/MissionManager/libMissionManager_arm64-v8a.a  src/PositionManager/libPositionManager_arm64-v8a.a  src/QtLocationPlugin/libQGCLocation_arm64-v8a.a  src/Settings/libSettings_arm64-v8a.a  src/Terrain/libTerrain_arm64-v8a.a  src/Vehicle/libVehicle_arm64-v8a.a  src/VehicleSetup/libVehicleSetup_arm64-v8a.a  src/Viewer3D/libViewer3D_arm64-v8a.a  src/VideoManager/libVideoManager_arm64-v8a.a  src/Geo/libGeo_arm64-v8a.a  src/FactSystem/FactControls/libFactControls_arm64-v8a.a  src/Utilities/Compression/libCompression_arm64-v8a.a  src/Comms/qtandroidserialport.build/libqtandroidserialport_arm64-v8a.a  src/Vehicle/Components/libVehicleComponents_arm64-v8a.a  src/FirmwarePlugin/APM/libAPMFirmwarePlugin_arm64-v8a.a  src/FirmwarePlugin/PX4/libPX4FirmwarePlugin_arm64-v8a.a  src/GPS/libGPSDrivers_arm64-v8a.a  src/Vehicle/Actuators/libVehicleActuators_arm64-v8a.a  src/MAVLink/LibEvents/libLibEventsWrapper_arm64-v8a.a  src/Vehicle/FactGroups/libVehicleFactGroups_arm64-v8a.a  src/AutoPilotPlugins/APM/libAPMAutoPilotPlugin_arm64-v8a.a  src/AutoPilotPlugins/PX4/libPX4AutoPilotPlugin_arm64-v8a.a  src/libQGC_arm64-v8a.a  src/QmlControls/libQmlControls_arm64-v8a.a  src/Utilities/libUtilities_arm64-v8a.a  src/API/libAPI_arm64-v8a.a  src/ADSB/libADSB_arm64-v8a.a  src/AnalyzeView/libAnalyzeView_arm64-v8a.a  src/Audio/libAudio_arm64-v8a.a  src/AutoPilotPlugins/libAutoPilotPlugins_arm64-v8a.a  src/Camera/libCamera_arm64-v8a.a  src/Comms/libComms_arm64-v8a.a  src/AutoPilotPlugins/Common/libCommonAutoPilotPlugin_arm64-v8a.a  src/FactSystem/libFactSystem_arm64-v8a.a  src/FirmwarePlugin/libFirmwarePlugin_arm64-v8a.a  src/FollowMe/libFollowMe_arm64-v8a.a  src/Gimbal/libGimbal_arm64-v8a.a  src/GPS/libGPS_arm64-v8a.a  src/VideoManager/VideoReceiver/GStreamer/libGStreamerReceiver_arm64-v8a.a  src/Joystick/libJoystick_arm64-v8a.a  src/MAVLink/libMAVLink_arm64-v8a.a  src/MissionManager/libMissionManager_arm64-v8a.a  src/PositionManager/libPositionManager_arm64-v8a.a  src/QtLocationPlugin/libQGCLocation_arm64-v8a.a  src/Settings/libSettings_arm64-v8a.a  src/Terrain/libTerrain_arm64-v8a.a  src/Vehicle/libVehicle_arm64-v8a.a  src/VehicleSetup/libVehicleSetup_arm64-v8a.a  src/Viewer3D/libViewer3D_arm64-v8a.a  src/VideoManager/libVideoManager_arm64-v8a.a  src/Geo/libGeo_arm64-v8a.a  src/FactSystem/FactControls/libFactControls_arm64-v8a.a  src/Utilities/Compression/libCompression_arm64-v8a.a  src/Comms/qtandroidserialport.build/libqtandroidserialport_arm64-v8a.a  src/Vehicle/Components/libVehicleComponents_arm64-v8a.a  s [... cut ...]
>>>               gstavvidenc.c.o:(gst_ffmpegvidenc_send_frame) in archive _deps/gstreamer/install/gstreamer/arm64/lib/gstreamer-1.0/libgstlibav.a
>>> referenced 2 more times

ld: error: undefined symbol: av_malloc_array
>>> referenced by gstavaudenc.c:459 (/home/nirbheek/projects/repos/cerbero.git/1.22/build/sources/android_universal/arm64/gst-libav-1.0-1.22.12/_builddir/../ext/libav/gstavaudenc.c:459)
>>>               gstavaudenc.c.o:(gst_ffmpegaudenc_send_frame) in archive _deps/gstreamer/install/gstreamer/arm64/lib/gstreamer-1.0/libgstlibav.a

the NDK version is 25.1.8937393.

some tips?
thanks

HTRamsey, 2024-10-18:
Are you building from linux? Make sure you have pkgconfig installed, as these should all be found automatically

david6096, 2024-10-18:
yes,ubuntu 20.04,and pkg-config works properly.

HTRamsey, 2024-10-18:
Are you on latest daily? For some reason your pkgconfig still has references to /home/nirbheek/projects/repos/cerbero.git/1.22/build/sources/ even though that should be overwritten

david6096, 2024-10-18:
yes,I'm sure that I'm on the latest daily

closed 2024-10-18 by HTRamsey

highfreq, 2024-11-21:
Have same prob, i  am on daily from yesterday:

ld: error: undefined symbol: g_module_symbol
>>> referenced by gstglcontext.c:918 (/home/nirbheek/projects/repos/cerbero.git/1.22/build/sources/android_universal/armv7/gst-plugins-base-1.0-1.22.12/_builddir/../gst-libs/gst/gl/gstglcontext.c:918)

any hint?

HTRamsey, 2024-11-21:
@highfreq can you show your configure output

highfreq, 2024-11-21:
Sure!!! If you cn tell me how to do it :)

Rohitnchan, 2024-11-21:
Hey folks,

I am getting similar error:

Bsymbolic -shared  -o libQGroundControl_armeabi-v7a.so @CMakeFiles\QGroundControl.rsp && cd ."
ld: error: undefined symbol: graphene_vec2_get_type
>>> referenced by gstglfiltershader.c:366 (/home/nirbheek/projects/repos/cerbero.git/1.22/build/sources/android_universal/armv7/gst-plugins-base-1.0-1.22.12/_builddir/../ext/gl/gstglfiltershader.c:366)
>>>               gstglfiltershader.c.o:(_set_uniform) in archive _deps/gstreamer/install/gstreamer/armv7/lib/gstreamer-1.0/libgstopengl.a
>>> referenced by gstglfiltershader.c:366 (/home/nirbheek/projects/repos/cerbero.git/1.22/build/sources/android_universal/armv7/gst-plugins-base-1.0-1.22.12/_builddir/../ext/gl/gstglfiltershader.c:366)
>>>               gstglfiltershader.c.o:(_set_uniform) in archive _deps/gstreamer/install/gstreamer/armv7/lib/gstreamer-1.0/libgstopengl.a

Can anyone make sense of this? been scratching my head with this one.

HTRamsey, 2024-11-21:
@Rohitnchan I'm not certain how some of you are getting results like this. Can you provide your configuration output as well?

Rohitnchan, 2024-11-21:
Hi @HTRamsey , Sorry, I am new to Qt. Can you please elaborate on the 'configuration output' comment

HTRamsey, 2024-11-21:
It'll look something like this and shows that gstreamer was setup correctly. It's output in QtCreator's General Messages.
```
[cmake] -- QGC_ENABLE_GST_VIDEOSTREAMING="ON"
[cmake] -- GSTREAMER_PREFIX="/home/jonglaser/Dev/projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_arm64_v8a-Debug/_deps/gstreamer/install/gstreamer/arm64"
[cmake] -- GSTREAMER_LIB_PATH="/home/jonglaser/Dev/projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_arm64_v8a-Debug/_deps/gstreamer/install/gstreamer/arm64/lib"
[cmake] -- Found GLIB2: /home/jonglaser/Dev/projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_arm64_v8a-Debug/_deps/gstreamer/install/gstreamer/arm64/lib/libglib-2.0.a   
[cmake] -- Checking for module 'gobject-2.0'
[cmake] --   Found gobject-2.0, version 2.74.4
[cmake] -- PKG_CONFIG_PATH 
[cmake] -- PKG_CONFIG_LIBDIR /home/jonglaser/Dev/projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_arm64_v8a-Debug/_deps/gstreamer/install/gstreamer/arm64/lib/pkgconfig:/home/jonglaser/Dev/projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_arm64_v8a-Debug/_deps/gstreamer/install/gstreamer/arm64/lib/gstreamer-1.0/pkgconfig
[cmake] -- PKG_CONFIG_EXECUTABLE="/usr/bin/pkg-config" ; PKG_CONFIG_ARGN="--static;--dont-define-prefix;--define-variable=prefix=/home/jonglaser/Dev/projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_arm64_v8a-Debug/_deps/gstreamer/install/gstreamer/arm64;--define-variable=libdir=/home/jonglaser/Dev/projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_arm64_v8a-Debug/_deps/gstreamer/install/gstreamer/arm64/lib;--define-variable=includedir=/home/jonglaser/Dev/projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_arm64_v8a-Debug/_deps/gstreamer/install/gstreamer/arm64/include"
[cmake] -- Checking for module 'gstreamer-1.0'
[cmake] --   Found gstreamer-1.0, version 1.22.12
[cmake] -- GStreamer_VERSION="1.22.12"
[cmake] -- GST_VERSION_MAJOR="1" ; GST_VERSION_MINOR="22" ; GST_VERSION_PATCH="12"
[cmake] -- GST_PLUGINS_VERSION="1.22.12"
[cmake] -- Checking for module 'gstreamer-1.0'
[cmake] --   Found gstreamer-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-base-1.0'
[cmake] --   Found gstreamer-base-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-video-1.0'
[cmake] --   Found gstreamer-video-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-gl-1.0'
[cmake] --   Found gstreamer-gl-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-allocators-1.0'
[cmake] --   Found gstreamer-allocators-1.0, version 1.22.12
[cmake] -- Checking for module 'libdrm'
[cmake] --   Package 'libdrm', required by 'virtual:world', not found
[cmake] -- Checking for module 'gstreamer-app-1.0'
[cmake] --   Found gstreamer-app-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-audio-1.0'
[cmake] --   Found gstreamer-audio-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-codecparsers-1.0'
[cmake] --   Found gstreamer-codecparsers-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-controller-1.0'
[cmake] --   Found gstreamer-controller-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-fft-1.0'
[cmake] --   Found gstreamer-fft-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-gl-prototypes-1.0'
[cmake] --   Found gstreamer-gl-prototypes-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-gl-wayland-1.0'
[cmake] --   Package 'gstreamer-gl-wayland-1.0', required by 'virtual:world', not found
[cmake] -- Checking for module 'gstreamer-gl-x11-1.0'
[cmake] --   Package 'gstreamer-gl-x11-1.0', required by 'virtual:world', not found
[cmake] -- Checking for module 'gstreamer-mpegts-1.0'
[cmake] --   Found gstreamer-mpegts-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-net-1.0'
[cmake] --   Found gstreamer-net-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-pbutils-1.0'
[cmake] --   Found gstreamer-pbutils-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-photography-1.0'
[cmake] --   Found gstreamer-photography-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-riff-1.0'
[cmake] --   Found gstreamer-riff-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-rtp-1.0'
[cmake] --   Found gstreamer-rtp-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-sdp-1.0'
[cmake] --   Found gstreamer-sdp-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-rtsp-1.0'
[cmake] --   Found gstreamer-rtsp-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-tag-1.0'
[cmake] --   Found gstreamer-tag-1.0, version 1.22.12
[cmake] -- Checking for module 'gstreamer-va-1.0'
[cmake] --   Package 'gstreamer-va-1.0', required by 'virtual:world', not found
[cmake] -- Found GStreamer: TRUE  found components: Allocators App Audio Codecparsers Controller Fft GlPrototypes Mpegts Net Pbutils Photography Riff Rtp Rtsp Sdp Tag missing components: GlEGL GlWayland GlX11 Va
[cmake] -- GSTREAMER_PLUGIN_PATH="/home/jonglaser/Dev/projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_arm64_v8a-Debug/_deps/gstreamer/install/gstreamer/arm64/lib/gstreamer-1.0"
[cmake] -- plugin="gstcoreelements"
[cmake] -- plugin="gstisomp4"
[cmake] -- plugin="gstlibav"
[cmake] -- plugin="gstmatroska"
[cmake] -- plugin="gstmpegtsdemux"
[cmake] -- plugin="gstopengl"
[cmake] -- plugin="gstplayback"
[cmake] -- plugin="gstrtp"
[cmake] -- plugin="gstrtpmanager"
[cmake] -- plugin="gstrtsp"
[cmake] -- plugin="gstsdpelem"
[cmake] -- plugin="gsttcp"
[cmake] -- plugin="gstudp"
[cmake] -- plugin="gstvideoparsersbad"
[cmake] -- plugin="gstx264"
[cmake] -- plugin="gstapp"
[cmake] -- plugin="gstandroidmedia"
[cmake] -- QGC_GST_QT6_PLUGIN_PATH="/home/jonglaser/Dev/projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_arm64_v8a-Debug/_deps/gstreamer_good_plugins-src/ext/qt6"
```

Rohitnchan, 2024-11-22:
Here it is 

[cmake] Running C:\Qt\Tools\CMake_64\bin\cmake.exe -S C:/Qt_Projects/qgroundcontrol -B C:/Qt_Projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_armeabi_v7a-Release "-DANDROID_PLATFORM:STRING=android-23" "-DANDROID_NDK:PATH=C:/Users/Voltin3/AppData/Local/Android/Sdk/ndk/25.1.8937393" "-DCMAKE_CXX_FLAGS_INIT:STRING=" "-DCMAKE_BUILD_TYPE:STRING=Release" "-DCMAKE_PROJECT_INCLUDE_BEFORE:FILEPATH=C:\Qt_Projects\qgroundcontrol\build\Android_Qt_6_6_3_Clang_armeabi_v7a-Release/.qtc/package-manager/auto-setup.cmake" "-DANDROID_STL:STRING=c++_shared" "-DCMAKE_PREFIX_PATH:PATH=C:/Qt/6.6.3/android_armv7" "-DANDROID_ABI:STRING=armeabi-v7a" "-DQT_NO_GLOBAL_APK_TARGET_PART_OF_ALL:BOOL=ON" "-DQT_QMAKE_EXECUTABLE:FILEPATH=C:/Qt/6.6.3/android_armv7/bin/qmake.bat" "-DANDROID_USE_LEGACY_TOOLCHAIN_FILE:BOOL=OFF" "-DANDROID_SDK_ROOT:PATH=C:/Users/Voltin3/AppData/Local/Android/Sdk" "-DCMAKE_GENERATOR:STRING=Ninja" "-DCMAKE_CXX_COMPILER:FILEPATH=C:/Users/Voltin3/AppData/Local/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/bin/clang++.exe" "-DCMAKE_TOOLCHAIN_FILE:FILEPATH=C:/Users/Voltin3/AppData/Local/Android/Sdk/ndk/25.1.8937393/build/cmake/android.toolchain.cmake" "-DCMAKE_C_COMPILER:FILEPATH=C:/Users/Voltin3/AppData/Local/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/bin/clang.exe" "-DQT_HOST_PATH:PATH=C:/Qt/6.6.3/mingw_64" "-DCMAKE_FIND_ROOT_PATH:PATH=C:/Qt/6.6.3/android_armv7" in C:\Qt_Projects\qgroundcontrol\build\Android_Qt_6_6_3_Clang_armeabi_v7a-Release.
[cmake] -- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.46.2.windows.1")
[cmake] -- QGC_GIT_BRANCH="master"
[cmake] -- QGC_GIT_HASH="acecadc9b"
[cmake] -- QGC_APP_VERSION_STR="latest-67-gacecadc9b"
[cmake] -- QGC_APP_VERSION="v4.3.0"
[cmake] -- QGC_APP_DATE="2023-11-20T18:46:48-08:00"
[cmake] -- QGC_APP_VERSION_VALID="0"
[cmake] -- QGC_APP_VERSION="4.3.0" ; QGC_APP_VERSION_MAJOR="4" ; QGC_APP_VERSION_MINOR="3" ; QGC_APP_VERSION_PATCH="0"
[cmake] -- Android: Targeting API '23' with architecture 'arm', ABI 'armeabi-v7a', and processor 'armv7-a'
[cmake] -- Android: Selected unified Clang toolchain
[cmake] -- The C compiler identification is Clang 14.0.6
[cmake] -- The CXX compiler identification is Clang 14.0.6
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: C:/Users/Voltin3/AppData/Local/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/bin/clang.exe - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: C:/Users/Voltin3/AppData/Local/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/bin/clang++.exe - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- QT_ANDROID_APPLICATION_ARGUMENTS="" ; QT_HOST_PATH="C:/Qt/6.6.3/mingw_64"
[cmake] -- QT_ANDROID_MULTI_ABI_FORWARD_VARS="QGC_STABLE_BUILD;QT_HOST_PATH"
[cmake] -- Signing APK
[cmake] -- QT_ANDROID_KEYSTORE_PATH 
[cmake] -- QT_ANDROID_KEYSTORE_ALIAS 
[cmake] -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[cmake] -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
[cmake] -- Check if compiler accepts -pthread
[cmake] -- Check if compiler accepts -pthread - yes
[cmake] -- Found Threads: TRUE
[cmake] -- Performing Test HAVE_STDATOMIC
[cmake] -- Performing Test HAVE_STDATOMIC - Success
[cmake] -- Found WrapAtomic: TRUE
[cmake] -- Performing Test HAVE_EGL
[cmake] -- Performing Test HAVE_EGL - Success
[cmake] -- Found EGL: C:/Users/Voltin3/AppData/Local/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include (found version "1.5")
[cmake] -- Performing Test HAVE_GLESv2
[cmake] -- Performing Test HAVE_GLESv2 - Success
[cmake] -- Found GLESv2: C:/Users/Voltin3/AppData/Local/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include
[cmake] -- Found WrapVulkanHeaders: C:/Users/Voltin3/AppData/Local/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include
[cmake] -- Could NOT find Qt6WaylandClient (missing: Qt6WaylandClient_DIR)
[cmake] CMake Warning at C:/Qt/6.6.3/android_armv7/lib/cmake/Qt6/Qt6Config.cmake:221 (message):
[cmake]   Failed to find optional Qt component "WaylandClient".
[cmake] 
[cmake]   Expected Config file at
[cmake]   "C:/Qt/6.6.3/android_armv7/lib/cmake/Qt6WaylandClient/Qt6WaylandClientConfig.cmake"
[cmake]   does NOT exist
[cmake] 
[cmake] Call Stack (most recent call first):
[cmake]   CMakeLists.txt:157 (find_package)
[cmake] 
[cmake] 
[cmake] -- Building Exiv2
[cmake] -- Found Python3: C:/Users/Voltin3/AppData/Local/Programs/Python/Python39/python.exe (found version "3.9.9") found components: Interpreter
[cmake] -- Looking for C++ include filesystem
[cmake] CMake Warning (dev) at C:/Qt/Tools/CMake_64/share/cmake-3.29/Modules/CheckIncludeFileCXX.cmake:76 (message):
[cmake]   Policy CMP0075 is not set: Include file check macros honor
[cmake]   CMAKE_REQUIRED_LIBRARIES.  Run "cmake --help-policy CMP0075" for policy
[cmake]   details.  Use the cmake_policy command to set the policy and suppress this
[cmake]   warning.
[cmake] 
[cmake]   CMAKE_REQUIRED_LIBRARIES is set to:
[cmake] 
[... cut ...]
[cmake]   processed, set CMP0100 to NEW.  If any of the files should not be
[cmake]   processed, explicitly exclude them by setting the source file property
[cmake]   SKIP_AUTOGEN:
[cmake] 
[cmake]     set_property(SOURCE file.hh PROPERTY SKIP_AUTOGEN ON)
[cmake] 
[cmake] This warning is for project developers.  Use -Wno-dev to suppress it.
[cmake] 
[cmake] -- Generating done (2.5s)
[cmake] -- Build files have been written to: C:/Qt_Projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_armeabi_v7a-Release
[cmake] 
[cmake] Elapsed time: 02:58.

HTRamsey, 2024-11-22:
It's not immediately clear that anything is wrong with your configuration stage. We do have a windows-android autobuild that's passing right now, but I will try a few things on a local machine to see if I can't find anything that would cause this.

Rohitnchan, 2024-11-22:
I currently have Gstreamer v1.24.9 on the windows machine while the Gsteamer android downloaded on the Qt is v1.22.12

Do you reckon the version mismatch might be causing this?

Rohitnchan, 2024-11-22:
Here the entire compile output if it helps with the investigation 

10:06:00: Running steps for project QGroundControl...
10:06:00: Starting: "C:\Qt\Tools\CMake_64\bin\cmake.exe" --build C:/Qt_Projects/qgroundcontrol/build/Android_Qt_6_6_3_Clang_armeabi_v7a-Release --target all
[1/783 2.1/sec] Automatic MOC and UIC for target UTMSP
[2/783 4.0/sec] Automatic MOC and UIC for target ulog_cpp
[3/783 5.9/sec] Automatic MOC and UIC for target AirLink
[4/783 7.7/sec] Automatic MOC and UIC for target exiv2lib_int
[5/783 9.6/sec] Automatic MOC and UIC for target xz
[6/783 11.3/sec] Automatic MOC and UIC for target MockLink
[7/783 13.0/sec] Automatic MOC and UIC for target exiv2lib
[8/783 14.5/sec] Automatic MOC and UIC for target zlibstatic
[9/783 16.2/sec] Automatic MOC and UIC for target GPSDrivers
[10/783 17.2/sec] Building CXX object _deps/ulogparser-build/ulog_cpp/CMakeFiles/ulog_cpp.dir/ulog_cpp_autogen/mocs_compilation.cpp.o
[11/783 18.7/sec] Automatic MOC and UIC for target QtMultimediaReceiver
[12/783 20.2/sec] Generating qrc_utmsp_dummy.cpp
[13/783 21.5/sec] Automatic MOC and UIC for target GeographicLib_STATIC
[14/783 22.6/sec] Automatic MOC and UIC for target shp
[15/783 22.8/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/exiv2lib_int_autogen/mocs_compilation.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[16/783 7.6/sec] Building CXX object _deps/ulogparser-build/ulog_cpp/CMakeFiles/ulog_cpp.dir/writer.cpp.o
[17/783 8.0/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/image_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[18/783 8.4/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/helper_functions.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[19/783 8.8/sec] Building CXX object _deps/ulogparser-build/ulog_cpp/CMakeFiles/ulog_cpp.dir/reader.cpp.o
[20/783 9.2/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/cr2header_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[21/783 7.6/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/casiomn_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[22/783 7.9/sec] Automatic MOC and UIC for target VideoReceiver
[23/783 8.2/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/fujimn_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[24/783 7.6/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/jp2image_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[25/783 6.0/sec] Building CXX object _deps/ulogparser-build/ulog_cpp/CMakeFiles/ulog_cpp.dir/data_container.cpp.o
[26/783 6.2/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/crwimage_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[27/783 6.2/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/orfimage_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[28/783 6.0/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/makernote_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[29/783 6.2/sec] Building CXX object _deps/ulogparser-build/ulog_cpp/CMakeFiles/ulog_cpp.dir/messages.cpp.o
[30/783 6.2/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/rw2image_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[31/783 6.4/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/panasonicmn_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[32/783 6.5/sec] Building CXX object _deps/ulogparser-build/ulog_cpp/CMakeFiles/ulog_cpp.dir/simple_writer.cpp.o
[33/783 6.6/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/minoltamn_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[34/783 6.7/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/pentaxmn_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[35/783 6.8/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib.dir/exiv2lib_autogen/mocs_compilation.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[36/783 7.0/sec] Automatic MOC and UIC for target gstqml6gl
[37/783 7.0/sec] Building CXX object _deps/exiv2-build/src/CMakeFiles/exiv2lib_int.dir/olympusmn_int.cpp.o
clang++: warning: argument unused during compilation: '-fstack-clash-protection' [-Wunused-command-line-argument]
[... cut ...]
>>>               gstgltransformation.c.o:(gst_gl_transformation_src_event) in archive _deps/gstreamer/install/gstreamer/armv7/lib/gstreamer-1.0/libgstopengl.a

ld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
10:12:08: The process "C:\Qt\Tools\CMake_64\bin\cmake.exe" exited with code 1.
Error while building/deploying project QGroundControl (kit: Android Qt 6.6.3 Clang armeabi-v7a)
When executing step "Build"
10:12:08: Elapsed time: 06:08.

HTRamsey, 2024-11-22:
I think I might know what to do, I'll get you to try something later probably

Rohitnchan, 2024-11-22:
Roger.

HTRamsey, 2024-11-22:
Can you uncomment lines 659 & 660 in FindGStreamer.cmake?

Rohitnchan, 2024-11-22:
Clean Built it with the lines uncommented and the same error follows

HTRamsey, 2024-11-22:
Can you make it like this and try again?
```
pkg_check_modules(PC_graphene IMPORTED_TARGET graphene-1.0)
if(TARGET PkgConfig::PC_graphene)
    target_link_libraries(GStreamer::GStreamer INTERFACE PkgConfig::PC_graphene)
endif()

target_include_directories(GStreamer::GStreamer
    INTERFACE
        ${GSTREAMER_PREFIX}/include
        ${GSTREAMER_PREFIX}/include/glib-2.0
        ${GSTREAMER_PREFIX}/include/graphene-1.0
        ${GSTREAMER_PREFIX}/include/gstreamer-1.0
        ${GSTREAMER_LIB_PATH}/glib-2.0/include
        ${GSTREAMER_LIB_PATH}/graphene-1.0/include
        ${GSTREAMER_LIB_PATH}/gstreamer-1.0/include
)
```

Rohitnchan, 2024-11-24:
Hey @HTRamsey,

I tried the above step and it built itself but this time with a new bunch of errors

FAILED: libQGroundControl_armeabi-v7a.so 
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\Users\Voltin3\AppData\Local\Android\Sdk\ndk\25.1.8937393\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe --target=armv7-none-linux-androideabi23 --sysroot=C:/Users/Voltin3/AppData/Local/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/windows-x86_64/sysroot -fPIC -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -march=armv7-a -mthumb -Wformat -Werror=format-security -fexceptions -frtti -stdlib=libc++ -O3 -DNDEBUG  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--gc-sections -Qunused-arguments -Wl,--no-undefined   -Wl,-Bsymbolic -pthread -shared  -o libQGroundControl_armeabi-v7a.so @CMakeFiles\QGroundControl.rsp && cd ."
ld: error: undefined symbol: jpeg_CreateDecompress
>>> referenced by gstgloverlay.c:682 (/home/nirbheek/projects/repos/cerbero.git/1.22/build/sources/android_universal/armv7/gst-plugins-base-1.0-1.22.12/_builddir/../ext/gl/gstgloverlay.c:682)
>>>               gstgloverlay.c.o:(gst_gl_overlay_filter_texture) in archive _deps/gstreamer/install/gstreamer/armv7/lib/gstreamer-1.0/libgstopengl.a

ld: error: undefined symbol: jpeg_std_error
>>> referenced by gstgloverlay.c:683 (/home/nirbheek/projects/repos/cerbero.git/1.22/build/sources/android_universal/armv7/gst-plugins-base-1.0-1.22.12/_builddir/../ext/gl/gstgloverlay.c:683)
>>>               gstgloverlay.c.o:(gst_gl_overlay_filter_texture) in archive _deps/gstreamer/install/gstreamer/armv7/lib/gstreamer-1.0/libgstopengl.a

HTRamsey, 2024-11-24:
It seems like pkg-config isn't working correctly for whatever reason. I'll have to try building armv7 on a windows machine myself and see what happens. GStreamer can be a mess to work with, so I've been working on adding an alternative video receiver that doesn't require any extra dependencies. So for now you can disable gstreamer through the cmake option, until this is either fixed or the other video receiver is working.

highfreq, 2024-11-27:
For me, sadly, didn't make any diff.
After reading the new readme i uninstalled gstreamer from brew and reinstalled from links, but same problems.

i noticed i have the following on the cmake file, maybe could help diagnose the prob:

-DGST_PLUGIN_gstva_LIBRARY:FILEPATH=GST_PLUGIN_gstva_LIBRARY-NOTFOUND
-DGST_PLUGIN_gstvaapi_LIBRARY:FILEPATH=GST_PLUGIN_gstvaapi_LIBRARY-NOTFOUND

highfreq, 2024-11-27:
I am stuck, in QT6 i can't link gstreamer and QT5 isn't available on mac :)

HTRamsey, 2024-11-27:
Your best bet is to disable gstreamer for now

highfreq, 2024-11-27:
what can i use to see video in QGC than?

HTRamsey, 2024-11-27:
There's an alternative video receiver using QtMultimedia but it's not really finished yet

highfreq, 2024-11-27:
Ok, thank you for your help, i'll wait.

PR #12023 (2024-10-18): Linux: Improve GStreamer Dependencies
Fixes #12024