mavlink/qgroundcontrol #12041: Motor Test does nothing

twistedwings, 2024-10-28:
## Expected Behavior
Sliders spin motors

## Current Behavior
Sliders do not spin motors

## Steps to Reproduce:
Enter Motor Test
Move sliders

## System Information
- Operating System: Windows 10
- QGC Version: 4.3.0
- QGC build: daily
- Flight Controller: Matek H743
- Autopilot (with version): Ardupilot Copter 4.5.7

HTRamsey, 2024-10-29:
@twistedwings could you elaborate on the issue you experienced? I tried on both 4.3 and the current daily without issues

bytesByHarsh, 2024-10-29:
Hi @twistedwings I tried recreating this issue in both Daily and Stable_V4.3 branch using SITL and it seems to be working fine.

twistedwings, 2024-10-29:
I tested again with a different copter and a different flight controller.
Same thing, no motor spinning.
Motortest in Mission Planner works.

Matek F405-CTR
ArduCopter 4.5.7
QGC stable
QGC daily

DonLakeFlyer, 2024-11-15:
@twistedwings Can you try older firmwares to see if this is some firmware side change?

twistedwings, 2024-11-16:
I tested 4.0.0 and 4.5.5.
Same thing.

Hauk-Lykke, 2025-07-01:
I experience the same with Matek F405-CTR.

DonLakeFlyer, 2025-07-01:
@HTRamsey Do you have an ArduPilot vehicle to test this one. I don't have one right now.

HTRamsey, 2025-07-01:
I was never able to reproduce this previously, but I can try again

DonLakeFlyer, 2025-09-01:
@twistedwings @Hauk-Lykke Does this still repro in 5.0?