mavlink/qgroundcontrol #12054: Application Crashing

rzappe, 2024-10-30:
## Expected Behavior
The pop up window with the individual parameter from the error window  should pop up and be editable. Then save the parameter and close the popup window.

## Current Behavior
The pop up window with the individual parameter from the error window causes QGC to crash when you try and select the drop down to change the parameter.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. click on the QGC logo to see errors
2. Click on the suggested parameter to change for no RC behavior
3. Click on drop down to change parameter
4. QGC Crashes

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: 4.4.2 (Continuous Release)
- QGC build: daily
- Flight Controller: mRo Control Zero H7
- Autopilot (with version): Latest

## Detailed Description
n/a

## Log Files and Screenshots

HTRamsey, 2024-10-30:
I''m not sure I entirely follow what widget you are saying is causing the crash. Maybe you can provide a screenshot?

rzappe, 2024-10-30:
> I''m not sure I entirely follow what widget you are saying is causing the crash. Maybe you can provide a screenshot?

This is the menu referenced.

[image]
[image]

rzappe, 2024-10-31:
@HTRamsey So now on the latest daily build it won't crash and I can select the drop down, but you cannot click on any of the items in the drop down. Check boxes in other items down work either.