mavlink/qgroundcontrol #12057: SRT Support in Video Streaming

PrcsnFlyer, 2024-10-30:
Using the SRT protocol for receiving video feeds would be highly beneficial, especially since it is already supported by GStreamer, making implementation straightforward.

SRT offers several advantages over traditional protocols like MJPEG and RTSP, primarily through features like Forward Error Correction (FEC) and encryption. These enhancements improve reliability and security without adding significant latency or bandwidth overhead. 

For more information about SRT, please refer to the following resources:

[SRT on GitHub](https://github.com/Haivision/srt)
[SRT Support in GStreamer](https://gstreamer.freedesktop.org/documentation/srt/srtsrc.html?gi-language=c)

Paschalis, 2024-10-31:
I've implemented an example for H.265 video and audio encoding using the `gst-launch-1.0` terminal tool, leveraging the SRT protocol for streaming. Below are the details for both the sender and receiver pipelines, along with their respective diagrams.

## Sender Side Pipeline:

```bash
gst-launch-1.0 -e v4l2src device=/dev/video0 ! \
  queue max-size-buffers=1 ! "video/x-raw, width=640, height=480, framerate=30/1" ! \
  videoconvert ! clockoverlay time-format="8A, %d %B %y, H:%M:%S" ! \
  queue max-size-buffers=1 ! x265enc tune=zerolatency ! video/x-h265 ! mux. \
  alsasrc device="hw:1" ! audio/x-raw, width=16, depth=16, rate=44100, channels=1 ! \
  queue ! audioconvert ! audioresample ! fdkaacenc ! aacparse ! \
  mpegtsmux name=mux ! srtsink uri=srt://:8888/
```

## Receiver Side Pipeline:

```bash
gst-launch-1.0 srtsrc uri=srt://<sender-ip>:8888 ! \
  tsdemux name=demux ! \
  queue ! h265parse ! video/x-h265 ! avdec_h265 ! videoconvert ! autovideosink \
  demux. ! queue ! aacparse ! avdec_aac ! audioconvert ! autoaudiosink
```

## Diagrams:
- **Sender Pipeline Diagram**: 

[image]

- **Receiver Pipeline Diagram**: 
[image]

These pipelines allow the sender to stream H.265 video and audio using SRT, while the receiver decodes and plays back the content. If you're interested, I can also provide the C code for this implementation. Also, I will try to integrate the SRT protocol into QGroundControl

michelc33, 2025-08-17:
What is the status of srt support ? Not clear