mavlink/qgroundcontrol #12071: Interface Issue - Grey Screen and Non-functioning Buttons

M-BHS, 2024-11-06:
## Expected Behavior
A functioning interface and app.

## Current Behavior
The app is not properly working. It shows a grey screen and none of the buttons work so I can't even check the in-app logs or change settings.

## System Information
When posting bug reports, include the following information
- Operating System: MacOs Sequoia 15.1.0
- QGC Version: 4.2.2
- QGC build: Stable
- Autopilot (with version): PX4

## Detailed Description
Hello everyone,

One of my colleagues recently encountered a weird issue on his MacBook.
The app interface seems broken, the screen is grey instead of showing the map and he can't even open the menus or do anything, the buttons just don't do anything. The app was properly working a few weeks ago.

The app was installed via Brew in the first place, we fully uninstalled and reinstalled the app via Homebrew and also via the .dmg file, we even flushed the app contents etc. but the problem still persists.
On a similar Macbook (same chip, specs, etc.), everything works fine so I'm a bit lost.

Any clues on how we could get the app to work properly again ?

Thanks in advance

## Log Files and Screenshots

[image]

HTRamsey, 2024-11-06:
You could try deleting the config & cache data. Also MacOS 15 isn't officially supported on Qt5.15 https://doc.qt.io/qt-5/macos.html

Geko580, 2024-11-12:
I have the same problem, screen frozen, and no action possible... Mac OS Ventura 13.7.1 (I am trying to connect Pixhawk4, Arducopter). Before it was good on my iMac.

JacquesWelgemoed, 2024-11-14:
I had the same issue with QGroundControl version 4.4.2 on MacOS 15.0.1. 
I was able to solve it by running it from the terminal with the --clear-settings argument.

You can find the executable by navigating to your applications directory. Usually this would be 2 directories above your home directory. So in a new terminal:
 ```bash
 cd ../../Applications/QGroundControl.app/Contents/MacOS
 ```
 Then you can run QGroundControl as follows:
 ```bash
 ./QGroundControl --clear-settings
 ```
 
 Hope this helps!

Geko580, 2024-11-15:
Hello !

Very good ! It’s running again 

Thank you very much for your help

Best regards 

> 
> 
> I had the same issue with QGroundControl version 4.4.2 on MacOS 15.0.1.
> I was able to solve it by running it from the terminal with the --clear-settings argument.
> 
> You can find the executable by navigating to your applications directory. Usually this would be 2 directories above your home directory. So in a new terminal:
> 
> cd ../../Applications/QGroundControl.app/Contents/MacOS
> Then you can run QGroundControl as follows:
> 
> ./QGroundControl --clear-settings
> Hope this helps!

Gerard Pelletier
06 19 07 26 21

DonLakeFlyer, 2024-11-15:
I'm going to close this. I tried to repro but couldn't. If anyone else hits this please re-open this issue and don't do a --clear-settings until we can take a look again at it.

closed 2024-11-15 by DonLakeFlyer