mavlink/qgroundcontrol #12076: QGC 4.4.2 + AP4.5.7 Waypoint and Rally Point Read Write Issues

menschel, 2024-11-08:
## Expected Behavior
Waypoint (read from | write to) FC is possible.

## Current Behavior
During initial connection GQC reads configuration from FC.
After parameters are finished, it tries to continue with Mission Items, i.e. Waypoints, Rally Points.
It fails with error popup 
`Mission read failed, maximum retries exceeded.`
once for waypoints and another one for rally points.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Connect an FC with Arduplane 4.5.7 to GQC.
2. Wait for Configuration to load.
3. Expect Error Popups half-way through.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu Mate 24.04LTS
- QGC Version: 4.4.2
- QGC build: stable
- Flight Controller: SpeedyBeeF405WingMini
- Autopilot (with version): Arduplane 4.5.7

## Detailed Description
FC is connected by UDP Port 14550.
That port is fed from a RM BANDIT MICRO module attached to a RM TX12MK2.
RC connection runs mavlink via ELRS feature.
https://www.expresslrs.org/software/mavlink/

Running APM-Planner_2-2.0.30-rc3-x86_64 instead of GQC shows that there is some index issue in the transmitted waypoints.
I think the format has changed somehow.

## Log Files and Screenshots
```
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanMasterController.cc:225 - "PlanMasterController::loadFromVehicle calling _missionController.loadFromVehicle"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:130 - ""loadFromVehicle T:Mission read sequence""
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:964 - "_setTransactionInProgress "T:Mission" 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:146 - ""_requestList T:Mission _planType:_retryCount" 0 0"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:187 - ""Retrying T:Mission REQUEST_LIST retry Count" 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:146 - ""_requestList T:Mission _planType:_retryCount" 0 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:331 - ""_handleMissionCount T:Mission count:" 25"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Mission sequenceNumber:retry" 0 0"
[E] at :0 - "QString::arg: Argument missing: T:Mission, MISSION_ITEM"
[E] at :0 - "QString::arg: Argument missing: T:Mission, MISSION_COUNT"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:284 - ""Out of sequence ack T:Mission expected:received %2:%3""
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Mission MISSION_REQUEST retry Count" 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Mission sequenceNumber:retry" 0 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:431 - ""_handleMissionItem T:Mission seq:command:current:ardupilotHomePositionUpdate" 0 16 false false"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Mission sequenceNumber:retry" 1 0"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Mission MISSION_REQUEST retry Count" 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Mission sequenceNumber:retry" 1 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Mission MISSION_REQUEST retry Count" 2"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Mission sequenceNumber:retry" 1 2"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Mission MISSION_REQUEST retry Count" 3"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Mission sequenceNumber:retry" 1 3"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Mission MISSION_REQUEST retry Count" 4"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Mission sequenceNumber:retry" 1 4"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Mission MISSION_REQUEST retry Count" 5"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Mission sequenceNumber:retry" 1 5"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Mission MISSION_REQUEST retry Count" 6"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Mission sequenceNumber:retry" 1 6"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:431 - ""_handleMissionItem T:Mission seq:command:current:ardupilotHomePositionUpdate" 0 16 false false"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:463 - ""_handleMissionItem T:Mission mission item received item index which was not requested, disregrarding:" 0"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:431 - ""_handleMissionItem T:Mission seq:command:current:ardupilotHomePositionUpdate" 1 22 false false"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Mission sequenceNumber:retry" 2 0"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:431 - ""_handleMissionItem T:Mission seq:command:current:ardupilotHomePositionUpdate" 1 22 false false"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:463 - ""_handleMissionItem T:Mission mission item received item index which was not requested, disregrarding:" 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:431 - ""_handleMissionItem T:Mission seq:command:current:ardupilotHomePositionUpdate" 1 22 false false"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:463 - ""_handleMissionItem T:Mission mission item received item index which was not requested, disregrarding:" 1"
[... 97 lines cut ...]
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:146 - ""_requestList T:Rally _planType:_retryCount" 2 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:323 - ""_handleMissionCount T:Rally Incorrect mission_type received expected:actual" 2 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:331 - ""_handleMissionCount T:Rally count:" 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Rally sequenceNumber:retry" 0 0"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Rally MISSION_REQUEST retry Count" 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Rally sequenceNumber:retry" 0 1"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Rally MISSION_REQUEST retry Count" 2"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Rally sequenceNumber:retry" 0 2"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Rally MISSION_REQUEST retry Count" 3"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Rally sequenceNumber:retry" 0 3"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Rally MISSION_REQUEST retry Count" 4"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Rally sequenceNumber:retry" 0 4"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:198 - ""Retrying T:Rally MISSION_REQUEST retry Count" 5"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:354 - ""_requestNextMissionItem T:Rally sequenceNumber:retry" 0 5"
[... cut ...]
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanManager.cc:964 - "_setTransactionInProgress "T:Rally" 0"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanMasterController.cc:267 - "PlanMasterController::_loadRallyPointsComplete"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanMasterController.cc:267 - "PlanMasterController::_loadRallyPointsComplete"
[D] at ../../qgroundcontrol/qgroundcontrol/src/MissionManager/PlanMasterController.cc:267 - "PlanMasterController::_loadRallyPointsComplete"`
``

DonLakeFlyer, 2024-11-15:
This looks like a firmware side protocol screw up. That said the full PlanManager lof isn't quite right. I just fixed a bug in there to get better logging. It was just merged. A new daily build with that fix should be available in about an hour. Can you repro this on daily and get a new PlanManager log from that?

menschel, 2024-11-16:
Sure, I'll wait for tomorrow and test with whatever Binary is linked here for linux.
https://docs.qgroundcontrol.com/master/en/qgc-user-guide/releases/daily_builds.html

Thanks for looking into the issue.

menschel, 2024-11-17:
I tested daily build and I can relate the issue to the Mavlink via ELRS feature.

When I connect the FC via Serial / USB everything works.

I'm currently struggling to actually log something because the console log does not show anything that I presumably selected for logging.

menschel, 2024-11-17:
Sorry, can't get any logging working with daily build, even `--logging:full` does not display anything that I selected for logging. All I get are some tile warnings from `terrain-ce.suite.auterion.com`

I'm running the daily of this commit
https://github.com/mavlink/qgroundcontrol/commit/1b8f5037a7ce913ed36bcd16f8c39399fcea0639

But I'll try one last thing here, not going through port 14550 but through a serial attached to the elrs module's usb port.

menschel, 2024-11-17:
Same effect through the elrs modules usb port, but it was much much faster until it threw the error message.

Again absolutely no logging even to command line except for the tile warnings.

menschel, 2024-11-19:
Agreed,

I believe it has something to do with being a stubborn sender.
https://www.expresslrs.org/software/mavlink/#stubborn-sender

Closing the issue therefore. I'll try to achieve some logging and attach it here for reference.

menschel, 2024-11-21:
I have another indicator that it's related to ELRS. The same test bench setup with mLRS on Seeed WIO-E5 modules works without any issues, also 900Mhz RF Link. I will open an Issue there.

menschel, 2024-11-23:
@DonLakeFlyer Am I correct in the assumption that 
[_retryTimeoutMilliseconds = 250](https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/PlanManager.h#L73) can be increased for testing?
I believe this timeout is part of the issue on the slower 900Mhz link while it is sufficient on the 2G4 link.
I'll try to build it locally according to [guide](https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/).

DonLakeFlyer, 2024-12-05:
Sorry lost track of this. Yeah that seems short. Try changing it to 1500 to see what happens.

menschel, 2024-12-09:
Setting `_retryTimeoutMilliseconds = 1500` has solved the initial issue at hand. I'm now searching for the corresponding timeout while writing fences and rally points.

I'll reopen this issue and eventually push PR afterwards.

BTW: Good writeup on the building instructions, I was struggling with distribution based qt packages for a week until I finally gave in to using the online installer and create an account.

DonLakeFlyer, 2024-12-09:
> I'm now searching for the corresponding timeout while writing fences and rally points.

If I remember correctly PlanManager is the base class for all of those isn't it?

menschel, 2024-12-09:
Aye, everything was solved by tweaking the timeouts. I pushed PR. Thanks for bearing with me ^^;

closed 2025-02-02 by DonLakeFlyer

menschel, 2025-02-05:
When I have time, I'll send a patch via git send-email instead of a PR, 
now that github has thrown me out. I know where to add the latency 
detection code.

Cheers.