mavlink/qgroundcontrol #12084: gstreamer-1.0-devel-msvc-x86_64-1.18.1.msi

gremsyQuanLe, 2024-11-11:
how to download "gstreamer-1.0-devel-msvc-x86_64-1.18.1.msi" when deploying QGC. This version is not available on "https://gstreamer.freedesktop.org/data/pkg/windows/" if anyone has it please share.Thank you so much!

HTRamsey, 2024-11-11:
You can just download 1.18.6

gremsyQuanLe, 2024-11-11:
I know, but it doesn't get pipeline from
gst_parse_launch ("v4l2src device=/dev/video0 ! video/x-raw,width=640,height=512,format=GRAY16_LE,framerate=9/1 \
! videoconvert ! x264enc speed-preset=superfast tune=4 ! appsink name=vid drop=true", NULL);

I tested with comand line "rtsp://192.168.12.141:1234/test" but it doesn't read video stream with "formatt = GRAY16"

gremsyQuanLe, 2024-11-11:
[image]
However, with QGC version v4.3.0 it is displayed and according to the information, this QGC version is being built with "gstreamer-1.0-devel-msvc-x86_64-1.18.1"

closed 2024-11-14 by HTRamsey

martinezmanuelmr, 2026-09-23:
Hello,

We are currently evaluating Allied Vision's official gst-vmbsrc plugin for Windows (gst-vmbsrc-1.1.0-beta1-windows-x86_64) with an Alvium camera.

Unfortunately, the official GStreamer download site (gstreamer.freedesktop.org) is currently returning HTTP 503 errors, so we are unable to download the required MSVC x64 Runtime and Development installers.

Does anyone know:

- A working mirror for the GStreamer Windows MSVC x64 installers?
- The latest recommended installer version currently available?
- An alternative download source?

Thank you.