mavlink/qgroundcontrol #12088: How to fix the error when using Qt 5.15 with JDK 11.0.25?

aibietdau, 2024-11-12:
I am using Qt version 5.15, and when installing devices for Android, I encounter an error and cannot install it. I am using JDK version 11.0.25. I have tried to find a solution but have not been able to. Please help me.
[image]

HTRamsey, 2024-11-12:
You have to use the current beta2 for QtCreator 15.0.0.
14.0.2 has some issues

aibietdau, 2024-11-12:
> You have to use the current beta2 for QtCreator 15.0.0. 14.0.2 has some issues

Thanks for your support, but I tried installing version 15.0.0 beta2 but still can't fix this error, do you have any other solutions?

HTRamsey, 2024-11-12:
So you are unable to install like Platform SDK 34 or something?

aibietdau, 2024-11-13:
> Vậy là bạn không thể cài đặt Platform SDK 34 hay gì đó phải không?
yes with jdk 11 i cant install any sdk when i click on set up sdk

aibietdau, 2024-11-13:
> #11907

i tried but the result is not as i expected, if i use another jdk for stable_v4.4 is it ok, for example jdk17 version

aibietdau, 2024-11-14:
> Sure, whatever ends up working. But you may want to change your NDK version as you should be using NDK 25
i thank you for the suggestion to solve the problem before, but after fixing the above error, there is a problem, that is i want to build for 32 bit version but i can't find armeabi_v7a, arm64_v8, please give me a suggestion
[image]
[image]

HTRamsey, 2024-11-30:
Yes, that is correct for Qt 5.15. The v7 and v8 targets are in Qt 6

closed 2024-11-30 by HTRamsey