mavlink/qgroundcontrol #12090: Plan load file dialog is not seen - ubuntu 22.04, Stable V4.4

zdanek, 2024-11-12:
I'm using latest stable branch V4.4, Ubuntu 22.04. I've built it on my computer, with CMake.
The question is, does it work for you? For similar setup?

After clicking Open... in order to open plan file, open dialog is not seen, although is visible on windows list:

Click
[image]

nothing is opened, although open dialog is on windows list:
[image]

Mobile dialog version works, if I force it in QGCOptions
```
virtual bool    useMobileFileDialog             () const { return false;}
```

[image]

DonLakeFlyer, 2024-11-15:
Two things:
* Does using the pre-built 4.4 downloaded from the web site work for you? 
* Not sure what state the cmake build is in for 4.4. Cmake is only guaranteed to work in daily. I would try using qmake.

zdanek, 2024-11-19:
@DonLakeFlyer good point. I've checked with downloaded latest stable, 4.4.2, and it **works**.
I suspect my local config. Also, I use cmake to build, since I work on CLion. Until now, everything worked correctly.

@DonLakeFlyer thank you for support.
Sorry for confusion. I'm closing this issue.