mavlink/qgroundcontrol #12092: QGC build doesn't work with generated mavlink libraries

owen487, 2024-11-14:
Hello,

Before I explain I do want to add I am a beginner so I am not 100% sure if this is a bug or something that I am doing wrong, give it a read, let me know your thoughts, and if this post is better suited somewhere else please let me know.

I am trying to create a QGC build for a custom mavlink message,  so as per the documentation I have generated my own mavlink library and replaced the pre-generated C library. I tried this in a couple ways.

- Include my message in common or development and generate
- Create a new dialect with the relevant includes and set config to my custom dialect

I use the mavgenerate gui tool to create the libraries, and make sure all the xml files neded are re-generated and replaced appropriately. Given the lack of further explanation in the in the guides I would assume that this should simply just work, but the build fails with the same error every time.

This is the error I get

[image]

The main parts being the following

/home/ubuntu/Desktop/Development/Again/qgroundcontrol/src/MissionManager/PlanManager.cc: In member function ‘void PlanManager::_writeMissionCount()’:
/home/ubuntu/Desktop/Development/Again/qgroundcontrol/src/MissionManager/PlanManager.cc:117:54: error: too few arguments to function ‘uint16_t mavlink_msg_mission_count_pack_chan(uint8_t, uint8_t, uint8_t, mavlink_message_t*, uint8_t, uint8_t, uint16_t, uint8_t, uint32_t)’
  117 |                                             _planType);
      |                                                      ^

/home/ubuntu/Desktop/Development/Again/qgroundcontrol/libs/mavlink/include/mavlink/v2.0/all/../ardupilotmega/../common/./mavlink_msg_mission_count.h:172:24: note: declared here
  172 | static inline uint16_t mavlink_msg_mission_count_pack_chan(uint8_t system_id, uint8_t component_id, uint8_t chan,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/home/ubuntu/Desktop/Development/Again/qgroundcontrol/src/MissionManager/PlanManager.cc: In member function ‘void PlanManager::_readTransactionComplete()’:
/home/ubuntu/Desktop/Development/Again/qgroundcontrol/src/MissionManager/PlanManager.cc:306:52: error: too few arguments to function ‘uint16_t mavlink_msg_mission_ack_pack_chan(uint8_t, uint8_t, uint8_t, mavlink_message_t*, uint8_t, uint8_t, uint8_t, uint8_t, uint32_t)’
  306 |                                           _planType);
      |                                                    ^

The main part I am not understanding is that if I take out my custom message, but still generate the library myself instead of using the pre-built, I still get this error, I am fairly sure that the method for generating my libraries is correct but it does definitely point to it being the issue.

Method for generating libraries 
1) Open mavgenerate GUI
2) Select V2, and C
3) Select all.xml to or custom.xml (depending on what method, all if using include in common/development, custom if using custom dialect)

Additionally, my custom message does build in PX4 using the various different methods as well so I'm fairly confident that the message isn't the issue.

I am also able to build just fine when using the pre-built so my method of building is fine.

In case any of this is relevant
Ubuntu 20.04
Attempted on both QGC daily and 4.4 stable

Thanks for any help
Owen

HTRamsey, 2024-11-14:
The mavlink messages generating that error were recently updated so you need a very recent version of the mavlink headers for it to build

owen487, 2024-11-14:
Hi @HTRamsey 
Thanks for the response, that would make sense as to why its not building, but I don't understand how I'm not generating "recent" headers. I use the latest xml files and am fully up to date with the mavgenerate side. Am I misunderstanding something? And if I compare my generated headers to the pre-built they seem to be the same so its not making sense to me.

To test I downloaded the very latest common.xml (The one that was had a commit less than a day ago) and generated the headers for it, and Ive still come to the same error, im not sure how else to get more recent headers.

Sorry if any of this is trivial stuff, as I said im a beginner so Im just trying to learn how everything works still

Thanks

HTRamsey, 2024-11-14:
I need to make some improvements to better allow custom messages, but in the mean time you can look at this PR where I fixed the same problem you are having #11332

owen487, 2024-11-15:
Hi @HTRamsey, 
Thanks for sending that to me, I was able to build and my custom messages are streaming through nicely. 

I did come across another error, was an "easy" fix (I don't understand why it fixed it) so don't feel like you need to explain why it happened, but equally feel free to explain it for me :)

In VideoManager.cc, line 695 and 717 needed to be changed from "case VIDEO_STREAM_TYPE_MPEG_TS_H264" --> "case VIDEO_STREAM_TYPE_MPEG_TS" as the H264 name was not declared.

Not an important feature for me so not vital but was just curious as it was not an error I've had in my other builds with the pre-built library.

Thanks for the help
Owen

closed 2024-11-15 by owen487