mavlink/qgroundcontrol #12094: Display bounces around with mult-vehicle connection

JustThinkInC, 2024-11-14:
## Expected Behavior
Vehicles should be displayed normally. Selected vehicle should be a darker red. The image below (from this [video](https://youtu.be/UWsya46ZG4M?t=403)) is an example of working a multi-vehicle display: 
[image]

## Current Behavior
Rather than two vehicle icons, 1 icon that just bounces around all over the place. .gif below for reference.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Start SITL (in my case using ardupilot/arducopter + MAVProxy)
2. Create connection in QGC to each SITL instance (using TCP in my case)
3. Connect
4. Connection to each will succeed but icon will bounce between vehicles.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10, Ubuntu 20.04.6 LTS (running in a VM though)
- QGC Version: 4.4.2, 4.4.1, 4.4.0, 4.3.0, 3.5.0 (others not tested)
- QGC build: stable, and built from sources
- Flight Controller: SITL
- Autopilot (with version): Ardupilot Copter 4.2.0-dev

## Log Files and Screenshots
[image]

PS: there is another issue https://github.com/mavlink/qgroundcontrol/issues/10279 from 2022 that sounds like this but no one has responded to it.

DonLakeFlyer, 2024-11-14:
This happens when you have multiple vehicles with the same mavlink id. Give each vehicle a unique id.

DonLakeFlyer, 2024-11-14:
Reopen issue if it still happens with correct unique ids.

closed 2024-11-14 by DonLakeFlyer

JustThinkInC, 2024-11-15:
Thanks @DonLakeFlyer but this still happens after giving each a vehicle unique ID using the `SYSID_THISMAV` (per the [docs](https://mavlink.io/en/services/mavlink_id_assignment.html)) but still have the issue. Screenshot confirming each vehicle has a different ID:

[image]

JustThinkInC, 2024-11-15:
Upon investigating further it seems this is actually an issue with SITL (or the way I'm instantiating the instances). While the screenshot above shows instances having separate IDs, when I added a debug line in QGC to print out the vehicle id on connection, the IDs were the same.
However, running with the [swarm ](https://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#swarming-with-sitl) configuration of SITL resolves this. I am not sure why the first method of manually spawning different instances and setting the parameter manually didn't work. Leaving this comment in case someone else runs into this.

DonLakeFlyer, 2024-11-15:
I vaguely remember with ArduPilot SITL needing to start up the instances manually from the command line. Not using the normal python script. And specifying the individual vehicle ids on the command line.