mavlink/qgroundcontrol #12109: Build on Jetson Orin Nano, failed with /usr/lib/x86_64-linux-gnu

lida2003, 2024-11-19:
Why there is `/usr/lib/x86_64-linux-gnu` in the config? I'm stucked here.

----

## Expected Behavior
Sucessfully build

## Current Behavior
There is no QT 6.6.3 avaliable, just 6.7.0  6.7.1 6.7.2 6.7.3 6.8.0

## Steps to Reproduce:
Just followed guide

## System Information
When posting bug reports, include the following information
- Jetson Orin Nano
- QGC Version: latest version on git hub
- QGC build: self-built from source
- Flight Controller: No
- Autopilot (with version): No

## Detailed Description
Why there is `/usr/lib/x86_64-linux-gnu` in the config?
It should be `/usr/lib/aarch64-linux-gnu`, I never provide `x86_64` info. Any idea with the build script?

## Log Files and Screenshots
```
qgroundcontrol$ cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_PROCESSOR=aarch64 -DCMAKE_LIBRARY_PATH=/usr/lib/aarch64-linux-gnu -DQt6_DIR=/opt/Qt/6.7.3/gcc_arm64/lib/cmake/Qt6/ -B build -G Ninja -DCMAKE_BUILD_TYPE=Debug
-- QGC_GIT_BRANCH="master"
-- QGC_GIT_HASH="1b8f5037a"
-- QGC_APP_VERSION_STR="latest-60-g1b8f5037a"
-- QGC_APP_VERSION="v4.3.0"
-- QGC_APP_DATE="2023-11-20T18:46:48-08:00"
-- QGC_APP_VERSION_VALID="0"
-- QGC_APP_VERSION="4.3.0" ; QGC_APP_VERSION_MAJOR="4" ; QGC_APP_VERSION_MINOR="3" ; QGC_APP_VERSION_PATCH="0"
-- Using CCache
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) (Required is at least version "2.3")
-- Checking for module 'GeographicLib>=2.3'
--   No package 'GeographicLib' found
-- Building GeographicLib
-- Joystick_targets="Joystick_resources_2;Joystickplugin_init"
-- Could NOT find SDL2: Found unsuitable version "2.0.10", but required is at least "2.30.0" (found /usr/lib/aarch64-linux-gnu/libSDL2.so)
-- Checking for module 'sdl2>=2.30.0'
--   Requested 'sdl2 >= 2.30.0' but version of sdl2 is 2.0.10
-- Building JoystickSDL
CMake Deprecation Warning at build/_deps/sdl-src/CMakeLists.txt:8 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at build/_deps/sdl-src/CMakeLists.txt:3245 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

--
-- SDL2 was configured with the following options:
--
-- Platform: Linux-5.10.216-tegra
-- 64-bit:   TRUE
-- Compiler: /usr/bin/cc
-- Revision: SDL-release-2.30.7-0-g9519b991
--
-- Subsystems:
--   Atomic:    ON
--   Audio:     OFF
[... 227 lines cut ...]

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in src/VideoManager/VideoReceiver/GStreamer/gstqml6gl/CMakeLists.txt:
  Imported target "GStreamer::GStreamer" includes non-existent path

    "/usr/lib/x86_64-linux-gnu/glib-2.0/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-- Generating done (2.7s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

HTRamsey, 2024-11-20:
The build system isn't adapted for embedded linux at the moment

lida2003, 2024-11-20:
Oh, really? I just got my jetson working. If the above issue is fixed, then it's perfectly.

[image]

HTRamsey, 2024-11-20:
In the mean time you can probably disable video and it should work

lida2003, 2024-11-20:
@HTRamsey I'll check if the video is OK or NOT.

BTW, Is there any embedded hardware avaliable for QGC?

lida2003, 2024-11-20:
@HTRamsey 

I checked rtp video, it didn't work. Does it has something to do with subsystem as below configured?

And how to enable it? Get those SDL* options on?

```
-- Subsystems:
--   Atomic:    ON
--   Audio:     OFF
--   Video:     OFF
--   Render:    OFF
--   Events:    ON
--   Joystick:  ON
--   Haptic:    ON
--   Hidapi:    ON
--   Power:     ON
--   Threads:   ON
--   Timers:    OFF
--   File:      OFF
--   Loadso:    ON
--   CPUinfo:   ON
--   Filesystem:        OFF
--   Sensor:    OFF
--   Locale:    OFF
--   Misc:      OFF
```

[image]

lida2003, 2024-11-20:
@HTRamsey Now I got all kind of warning message removed.

But there is still no video, is there anyway can help to debug the issue?

```
$ cmake -DQt6_DIR=/opt/Qt/6.7.3/gcc_arm64/lib/cmake/Qt6/ -B build -G Ninja -DCMAKE_BUILD_TYPE=Debug
-- Found Git: /usr/bin/git (found version "2.25.1")
-- QGC_GIT_BRANCH="master"
-- QGC_GIT_HASH="1b8f5037a"
-- QGC_APP_VERSION_STR="latest-60-g1b8f5037a"
-- QGC_APP_VERSION="v4.3.0"
-- QGC_APP_DATE="2023-11-20T18:46:48-08:00"
-- QGC_APP_VERSION_VALID="0"
-- QGC_APP_VERSION="4.3.0" ; QGC_APP_VERSION_MAJOR="4" ; QGC_APP_VERSION_MINOR="3" ; QGC_APP_VERSION_PATCH="0"
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CCache
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found WrapAtomic: TRUE
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so
-- Found WrapOpenGL: TRUE
-- Found XKB: /usr/lib/aarch64-linux-gnu/libxkbcommon.so (found suitable version "0.10.0", minimum required is "0.5.0")
-- Found WrapVulkanHeaders: /usr/include
-- Found Wayland_Client: /usr/lib/aarch64-linux-gnu/libwayland-client.so (found version "1.18.0")
-- Found Wayland_Server: /usr/lib/aarch64-linux-gnu/libwayland-server.so (found version "1.18.0")
-- Found Wayland_Cursor: /usr/lib/aarch64-linux-gnu/libwayland-cursor.so (found version "1.18.0")
-- Found Wayland_Egl: /usr/lib/aarch64-linux-gnu/libwayland-egl.so (found version "18.1.0")
-- Found Wayland: /usr/lib/aarch64-linux-gnu/libwayland-client.so;/usr/lib/aarch64-linux-gnu/libwayland-server.so;/usr/lib/aarch64-linux-gnu/libwayland-cursor.so;/usr/lib/aarch64-linux-gnu/libwayland-egl.so (found suitable version "1.18.0", minimum required is "1.15")
-- Found WaylandScanner: /usr/bin/wayland-scanner
-- Found LibExiv2: /usr/local/lib/libexiv2.so (found suitable version "0.28.3", minimum required is "0.28.2")
-- Checking for module 'speech-dispatcher'
--   Found speech-dispatcher, version 0.9.1
-- Performing Test HAVE_FLITE
-- Performing Test HAVE_FLITE - Success
-- Found Flite: /usr/lib/aarch64-linux-gnu/libflite.so
-- Found GeographicLib: /usr/local/lib (Required is at least version "2.3")
-- Found GeographicLib
-- GeographicLib_LIBRARIES="/usr/local/lib/libGeographicLib.so" ; GeographicLib_INCLUDE_DIRS="/usr/local/include" ; GeographicLib_LIBRARY_DIRS="/usr/local/lib"
-- Joystick_targets="Joystick_resources_2;Joystickplugin_init"
-- Found SDL2: /usr/local/lib/libSDL2.so (found suitable version "2.30.9", minimum required is "2.30.0")
-- Found SDL2main: /usr/local/lib/libSDL2main.a (found version "2.30.9")
-- Found JoystickSDL 2.30.9
-- Not enabling format target (clang-format not found)
-- Found ZLIB: /usr/local/lib/libz.a (found suitable version "1.3.1", minimum required is "1.3")
-- Found ZLIB 1.3.1
-- Found Shapelib: /usr/local/include (found suitable version "1.6.1", minimum required is "1.6.0")
-- Found Shapelib 1.6.1
-- UTMSP: Dummy is Initialized
-- QGC_ENABLE_GST_VIDEOSTREAMING="ON"
-- GSTREAMER_PREFIX="/usr"
-- GSTREAMER_LIB_PATH="/usr/lib/x86_64-linux-gnu"
-- Found GLIB2: /usr/lib/aarch64-linux-gnu/libglib-2.0.so
-- Checking for module 'gobject-2.0'
--   Found gobject-2.0, version 2.64.6
[... 93 lines cut ...]
-- MinSizeRel:
--
-- CMake Version:             3.31.0
-- CMake Prefix:
-- CMake Host System Name:    Linux
-- CMake Host System Version: 5.10.216-tegra
-- CMake System Name:         Linux
-- CMake System Version:      5.10.216-tegra
-- CMake Source Dir:          /home/daniel/Work/qgroundcontrol
-- CMake Toolchain File:
-- App Name:                  QGroundControl
-- Org Name:                  QGroundControl.org
-- App Domain:                org.qgroundcontrol
-- App Version:               latest-60-g1b8f5037a
-- Stable Build:              NO
-- Building Tests:            YES
-- Debug QML:                 NO
-- No Serial Links:           NO
-- Disable APM MAVLink:       NO
-- Build Dependencies:        NO
qmldir file not found at "/home/daniel/Work/qgroundcontrol/build/qml/QGroundControl"
-- Configuring done (38.2s)
-- Generating done (2.8s)
-- Build files have been written to: /home/daniel/Work/qgroundcontrol/build
```

lida2003, 2024-11-25:
I got below error when launch QGC. The video feed is H264, it works find on jetson video-viewer and other local network pc.

So the video source is OK.

```
$ ./build/QGroundControl
qgc.qgcapplication: Settings location "/home/daniel/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nqgc.*.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
VideoReceiverLog: GStreamer error: Your GStreamer installation is missing a plug-in.
Adding target QHostAddress("127.0.0.1") 52824
VideoReceiverLog: GStreamer error: Your GStreamer installation is missing a plug-in.
VideoReceiverLog: GStreamer error: Your GStreamer installation is missing a plug-in.
VideoReceiverLog: GStreamer error: Your GStreamer installation is missing a plug-in.
VideoReceiverLog: GStreamer error: Your GStreamer installation is missing a plug-in.
VideoReceiverLog: GStreamer error: Your GStreamer installation is missing a plug-in.
VideoReceiverLog: GStreamer error: Your GStreamer installation is missing a plug-in.
VideoReceiverLog: GStreamer error: Your GStreamer installation is missing a plug-in.
^CVideoReceiverLog: Stop called on empty URI
Exiting main
VideoReceiverLog: Stop called on empty URI
```

- And those video/audio plugin seems fine on jetson: such as gstreamer1.0-libav  /gstreamer1.0-plugins-bad / gstreamer1.0-plugins-good  etc. Please help to check if I missed some critical component.

```
$ aptitude search gstreamer1.0-
p   gstreamer1.0-adapter-pulseeffects                                           - GStreamer adapter plugin
i   gstreamer1.0-alsa                                                           - GStreamer plugin for ALSA
v   gstreamer1.0-audiosink                                                      -
v   gstreamer1.0-audiosource                                                    -
p   gstreamer1.0-autogain-pulseeffects                                          - GStreamer autogain plugin
i A gstreamer1.0-clutter-3.0                                                    - Clutter PLugin for GStreamer 1.0
p   gstreamer1.0-convolver-pulseeffects                                         - GStreamer convolver plugin
p   gstreamer1.0-crystalizer-pulseeffects                                       - GStreamer crystalizer plugin
p   gstreamer1.0-doc                                                            - GStreamer core documentation and manuals
p   gstreamer1.0-espeak                                                         - GStreamer plugin for eSpeak speech synthesis
i   gstreamer1.0-gl                                                             - GStreamer plugins for GL
i A gstreamer1.0-gtk3                                                           - GStreamer plugin for GTK+3
i   gstreamer1.0-libav                                                          - ffmpeg plugin for GStreamer
p   gstreamer1.0-libav-dbg                                                      - ffmpeg plugin for GStreamer (debug symbols)
p   gstreamer1.0-nice                                                           - ICE library (GStreamer plugin)
p   gstreamer1.0-omx-bellagio-config                                            - OpenMax plugins for GStreamer
p   gstreamer1.0-omx-generic                                                    - OpenMax plugins for GStreamer
p   gstreamer1.0-omx-generic-config                                             - OpenMax plugins for GStreamer
p   gstreamer1.0-opencv                                                         - GStreamer OpenCV plugins
i A gstreamer1.0-packagekit                                                     - GStreamer plugin to install codecs using PackageKit
p   gstreamer1.0-pipewire                                                       - GStreamer 1.0 plugin for the PipeWire multimedia server
i   gstreamer1.0-plugins-bad                                                    - GStreamer plugins from the "bad" set
p   gstreamer1.0-plugins-bad-dbg                                                - GStreamer plugins from the "bad" set (debug symbols)
p   gstreamer1.0-plugins-bad-doc                                                - GStreamer documentation for plugins from the "bad" set
v   gstreamer1.0-plugins-bad-faad                                               -
v   gstreamer1.0-plugins-bad-videoparsers                                       -
i   gstreamer1.0-plugins-base                                                   - GStreamer plugins from the "base" set
i A gstreamer1.0-plugins-base-apps                                              - GStreamer helper programs from the "base" set
p   gstreamer1.0-plugins-base-dbg                                               - GStreamer plugins from the "base" set
[... cut ...]
p   libffmpegthumbnailer-dev                                - development files for ffmpegthumbnailer
p   libffmpegthumbnailer4v5                                 - shared library for ffmpegthumbnailer
p   libspa-ffmpeg                                           - libraries for the PipeWire multimedia server - ffmpeg plugins
p   libxine2-ffmpeg                                         - MPEG-related plugins for libxine2
p   moc-ffmpeg-plugin                                       - ncurses based console audio player - ffmpeg plugin
```

lida2003, 2024-11-25:
Made a little bit of progress: https://github.com/mavlink/qgroundcontrol/issues/10638

```
$ ./build/QGroundControl --logging:VideoAllLog,GStreamerLog,GStreamerAPILog
qgc.qgcapplication: Settings location "/home/daniel/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nqgc.*.debug=false\nVideoAllLog.debug=true\nGStreamerLog.debug=true\nGStreamerAPILog.debug=true\nVideoManagerLog.debug=true\nVideoReceiverLog.debug=true\nGStreamerLog.debug=true\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
VideoReceiverLog: Starting "udp265://0.0.0.0:5600" , buffer -1
VideoReceiverLog: Started "udp265://0.0.0.0:5600"
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
VideoReceiverLog: Streaming started "udp265://0.0.0.0:5600"
Adding target QHostAddress("127.0.0.1") 52824
VideoReceiverLog: Starting decoding "udp265://0.0.0.0:5600"
VideoReceiverLog: Decoding started "udp265://0.0.0.0:5600"
VideoReceiverLog: _onNewDecoderPad "udp265://0.0.0.0:5600"
VideoReceiverLog: Unable to link video sink
VideoReceiverLog: _addVideoSink() failed
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 279
NvMMLiteBlockCreate : Block : BlockType = 279
```

lida2003, 2024-11-25:
Open a new ticket: [ "Unable to link video sink" on Jetson Orin Nano #12135 ](https://github.com/mavlink/qgroundcontrol/issues/12135)

closed 2024-11-25 by lida2003

Sheliahovskyi, 2024-12-13:
I have the same build issue with Gstreamer on the latest master branch on linux FEDORA40. 
CMAKE Log:
[cmake] CMake Error in src/VideoManager/VideoReceiver/GStreamer/gstqml6gl/CMakeLists.txt:
[cmake]   Imported target "GStreamer::GStreamer" includes non-existent path
[cmake] 
[cmake]     "/usr/lib/x86_64-linux-gnu/glib-2.0/include"

HTRamsey, 2024-12-13:
@Sheliahovskyi I don't use Fedora, so maybe the paths are different