mavlink/qgroundcontrol #12115: why when i try  build the qgroundcontrol i getted error i try build qgroundcontrol daily and i tried build the 4.4.2 but there are error why give me the steps to building 4.4.2 and what i need do to solve this error  here are error when i build qgroundcontrol4.4.2 D:\myproject\Qground\qgroundcontrol-4.4.2\CMakeLists.txt:25: error: By not providing "FindSDL2.cmake" in CMAKE_MODULE_PATH this project has asked CMake to find a package configuration file provided by "SDL2", but CMake did not find one.  Could not find a package configuration file provided by "SDL2" with any of the following names:    SDL2Config.cmake   sdl2-config.cmake  Add the installation prefix of "SDL2" to CMAKE_PREFIX_PATH or set "SDL2_DIR" to a directory containing one of the above files.  If "SDL2" provides a separate development package or SDK, be sure it has been installed. and the library is exist

abualiali, 2024-11-20:
Please only ask questions related to source code.
- For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources
- If your question still isn't answered, please check the forums: https://discuss.px4.io/c/qgroundcontrol/15.
- If it is about Qt or any QGC dependencies, please refer to them instead.

Argyraspides, 2024-11-23:
Sounds like you might not have SDL2 on your system. What operating system are you on? There are ``find_package()`` CMake commands which attempt to find SDL2 on your system automatically but if you don't have it installed in the default directories CMake tries to look in, or just don't have it at all, this is one of the errors it might throw.

If you're on Ubuntu or similar, you can just run the following on your command line:

```
sudo apt update
sudo apt install -y \
    libsdl2-dev libsdl2-2.0-0 \
    libsdl2-image-dev libsdl2-image-2.0-0 \
    libsdl2-mixer-dev libsdl2-mixer-2.0-0 \
    libsdl2-ttf-dev libsdl2-ttf-2.0-0 \
    libjpeg-dev libwebp-dev libtiff5-dev \
    libmikmod-dev libfishsound1-dev libsmpeg-dev \
    liboggz2-dev libflac-dev libfluidsynth-dev \
    libfreetype6-dev
```

Then try building again with CMake (I'm assuming this is what you used). Make sure to build from scratch by removing the ``build/`` folder in the root directory.

abualiali, 2024-11-24:
thank you i continue this problem

HTRamsey, 2024-11-30:
I would recommend not using cmake and use qmake instead for the older 4.X builds. But yes, it is related to not having sdl2, or at least it not being found

closed 2024-11-30 by HTRamsey