mavlink/qgroundcontrol #12116: Version error

highfreq, 2024-11-20:
I get the following error:

* What went wrong:
> Task :mapReleaseSourceSetPaths UP-TO-DATE
Execution failed for task ':checkReleaseAarMetadata'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
   > An issue was found when checking AAR metadata:
     
       1.  Dependency 'androidx.core:core:1.12.0' requires libraries and applications that
           depend on it to compile against version 34 or later of the
           Android APIs.
     
           : is currently compiled against android-33.
     
           Recommended action: Update this project to use a newer compileSdk
           of at least 34, for example 35.
     
           Note that updating a library or application's compileSdk (which
           allows newer APIs to be used) can be done separately from updating
           targetSdk (which opts the app in to new runtime behavior) and
           minSdk (which determines which devices the app can be installed
           on).

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

Any help would be greatly appreciated.

HTRamsey, 2024-11-20:
You can try reducing the version in the line `implementation 'androidx.core:core:1.12.0'` in build.gradle but really you need to install at least API 34 because 34 is required for apps on the play store as of August 2024

highfreq, 2024-11-20:
I installed 34 with android studio but i guess it is not how it should be done.

highfreq, 2024-11-21:
SOLVED!!!

Somehow i was pointing at the wrong directory. Once fixed all installed versions show up and selecting 35 it compiles fine.

Now gstreamer is my last problem i need to solve.

HTRamsey, 2024-11-21:
Check the configure output and it should look something like this for the gstreamer part:
[image]

highfreq, 2024-11-22:
Mine looks like this:

[cmake] -- QGC_ENABLE_GST_VIDEOSTREAMING="ON"
[cmake] -- GSTREAMER_PREFIX="/Users/corrado/qgroundcontrol/build/Android_Qt_6_6_3_Clang_armeabi_v7a-Release/_deps/gstreamer/install/gstreamer/armv7"
[cmake] -- GSTREAMER_LIB_PATH="/Users/corrado/qgroundcontrol/build/Android_Qt_6_6_3_Clang_armeabi_v7a-Release/_deps/gstreamer/install/gstreamer/armv7/lib"
[cmake] -- GStreamer_VERSION="1.22.12"
[cmake] -- GST_VERSION_MAJOR="1" ; GST_VERSION_MINOR="22" ; GST_VERSION_PATCH="12"
[cmake] -- GST_PLUGINS_VERSION="1.22.12"
[cmake] -- component="Core"
[cmake] -- component="Base"
[cmake] -- component="Video"
[cmake] -- component="Gl"
[cmake] -- component="Allocators"
[cmake] -- component="App"
[cmake] -- component="Audio"
[cmake] -- component="Codecparsers"
[cmake] -- component="Controller"
[cmake] -- component="Fft"
[cmake] -- component="Mpegts"
[cmake] -- component="Net"
[cmake] -- component="Pbutils"
[cmake] -- component="Photography"
[cmake] -- component="Riff"
[cmake] -- component="Rtp"
[cmake] -- component="Sdp"
[cmake] -- component="Rtsp"
[cmake] -- component="Tag"
[cmake] -- GSTREAMER_PLUGIN_PATH="/Users/corrado/qgroundcontrol/build/Android_Qt_6_6_3_Clang_armeabi_v7a-Release/_deps/gstreamer/install/gstreamer/armv7/lib/gstreamer-1.0"
[cmake] -- plugin="gstcoreelements"
[cmake] -- plugin="gstisomp4"
[cmake] -- plugin="gstlibav"
[cmake] -- plugin="gstmatroska"
[cmake] -- plugin="gstmpegtsdemux"
[cmake] -- plugin="gstopengl"
[cmake] -- plugin="gstplayback"
[cmake] -- plugin="gstrtp"
[cmake] -- plugin="gstrtpmanager"
[cmake] -- plugin="gstrtsp"
[cmake] -- plugin="gstsdpelem"
[cmake] -- plugin="gsttcp"
[cmake] -- plugin="gstudp"
[cmake] -- plugin="gstvideoparsersbad"
[cmake] -- plugin="gstx264"
[cmake] -- plugin="gstapp"
[cmake] -- plugin="gstandroidmedia"
[cmake] -- QGC_GST_QT6_PLUGIN_PATH="/Users/corrado/qgroundcontrol/build/Android_Qt_6_6_3_Clang_armeabi_v7a-Release/_deps/gstreamer_good_plugins-src/ext/qt6"
[cmake] -- QGC_ENABLE_QT_VIDEOSTREAMING="ON"

highfreq, 2024-11-22:
I see i do not have PKG in my output, should i enable it somewhere?

HTRamsey, 2024-11-22:
@highfreq It should automatically be getting pkg-config from gstreamer. Can you look at  #12124 and try that?

highfreq, 2024-11-23:
Made the mods in FindGStreamer.cmake and run cmake, if this is correct procedure than same error as before.

Thanks for helping me :)

highfreq, 2024-11-23:
Noticed that my configuration never says "Found gstreamer-1.0, version 1.22.12", is it ok?

highfreq, 2024-11-25:
Don't know if it makes any difference but i am trying to compile on Mac apple silicon.

HTRamsey, 2024-11-25:
I don't know a lot about Mac but that's currently the one OS I don't have a CI workflow for android set up to run. The windows and Linux ones seem to be working fine for android. Do you have pkg-config installed?

highfreq, 2024-11-25:
yes, i brew installed it.

Ho

HTRamsey, 2024-11-30:
Closing this as well since original issue was resolved

closed 2024-11-30 by HTRamsey