mavlink/qgroundcontrol #12127: QGroundControl Not connecting automatically, Major Issues

Flytking, 2024-11-23:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/support/support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Expected behavior: When I connect the USB cable from computer to the Pixhawk 6C, QGroundControl should automatically connect. When I turn on my radio transmitter, QGroundControl should allow me to calibrate the radio.

## Current Behavior
Current behavior: When I connect the USB cable from computer to the Pixhawk 6C, QGroundControl fails to connect. When I upgrade the firmware, QGroundControl connect for a moment but then it shows a communication error. See picture below. Also, when I turn on my radio transmitter, it fails to connect to QGroundControl. I know the radio transmitter is connected though because telemetry shows it is connected and the Receiver lights up green.

[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open QGroundControl on desktop
2. Connect USB cable from computer to Pixhawk 6C
3. Pixhawk 6C fails to connect
4. "Communication Lost" 
5. Cannot calibrate Radio or Sensors

## System Information
FC: Pixhawk 6C 
Radio: FrSky X10S Express 
Receiver: FrSky Archer GR8 
Ardupilot Firmware: ChibiOS, "Plane" Setting, PixHawk 6C - 4.5.7
QGC: Stable, v4.4.2

## Detailed Description
I have upgraded the firmware a dozen times just to see if there are any possible fixes. Firmware is the most updated possible. I am a beginner drone builder. I hope to learn as much as possible from the veterans here! Thank you!!

## Log Files and Screenshots
I don't know how to get the log files.

HTRamsey, 2024-11-30:
Can you try with the daily?

Flytking, 2024-11-30:
What is the daily?

HTRamsey, 2024-12-01:
https://github.com/mavlink/qgroundcontrol/releases/tag/latest

closed 2025-05-03 by HTRamsey