mavlink/qgroundcontrol #12132: Option to Hide "Camera Tools" in Fly View

itaru-fujitsuka, 2024-11-24:
**Feature Request:**  

I would like to request an option to hide the "Camera Tools" UI element in the Fly View of QGroundControl.  
[https://docs.qgroundcontrol.com/master/en/qgc-user-guide/fly_view/fly_view.html](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/fly_view/fly_view.html)

**Reasoning:**  
- Not all users require camera-related tools during their flight operations.  
- Hiding the "Camera Tools" can provide a cleaner and less cluttered interface for users focusing solely on flight monitoring and control.  
- This feature can be especially beneficial for setups without a connected camera or when the camera functionality is not in use.

**Proposed Solution:**  
- Add a toggle in the application settings (e.g., under *General Settings* or *Fly View Settings*) to enable or disable the "Camera Tools" UI element.  
- Alternatively, allow users to customize the Fly View layout to include or exclude specific tools.  

**Impact:**  
This change would improve the user experience for those who do not use camera features and prefer a simpler interface.  

Thank you for considering this request!

DonLakeFlyer, 2024-11-24:
If you vehicle doesn't report back mavlink protocol camera support QGC will show simple camera controls. You can turn these off from the "Show simple camera controls" App Settings. In Stable I think it's on the General tab. In Daily I believe it in the Fly View tab. Let me know if that doesn't work for you.

itaru-fujitsuka, 2024-11-27:
I tried that setting before asking the question, but it didn’t work.
Just to be sure, I tested several versions(4.4.2, 4.3.0, 4.2.8), restarted QGC, connected different vehicles, and even tried it on a different Windows PC today, but it still didn’t work.
What should I do?

DonLakeFlyer, 2024-11-27:
Did you try a daily build. Maybe it's broken in Stable.

itaru-fujitsuka, 2024-11-28:
I used the latest version.
The only thing that disappears is the photo and video switching.
I would like to turn off or minimize the recording button.

Also, the compass is in a different position, which is better than the original position. Can this be changed?

[image]
[image]

DonLakeFlyer, 2024-11-29:
Oh wait. Your Vehicle has video support. If your vehicle has video hardware which reports as available through mavlink there it no way to turn off video controls.