mavlink/qgroundcontrol #12136: QT5.15 For mac

highfreq, 2024-11-25:
Anybody knows how to get QT5.15 for Mac?

The online installer only shows QT6.

Thanks.

HTRamsey, 2024-11-25:
You have to check archive in the installer 
[image]

highfreq, 2024-11-25:
I did but on Mac it shows only starting from early QT6 versions, no QT5

HTRamsey, 2024-11-25:
Oh I have no idea then. I suspect as extended support for Qt5 is ending they may be removing it from their servers.

highfreq, 2024-11-25:
I read a bit around and looks like windows machies can see it but not mac.........

HTRamsey, 2024-11-25:
I think there's a way to add custom servers to the installer tool, maybe you can find one with the old downloads

HTRamsey, 2024-11-30:
I'm gonna close this, but I'll let you know when I make some progress on the video stuff

closed 2024-11-30 by HTRamsey